mne-tools/mne-python #7956: Ticks missing for small scale data in topomaps

datalw, 2020-07-03:
I am trying to plot power of different frequency bands using topomap with the same scale, so I can have an overview of comparison. But the topomaps with very small scale do not have complete colorbar ticks. Please see the figures below.
I do not want to leave only one colorbar in the figure due to layout reasons regarding some other figures. Does anyone know how I can let the ticks show correctly for every topomap? Thanks. 
P.S. The version of my mne was updated yesterday (master version) and the version of matplotlib is 3.2.2. 

[image]
[image]

Reproducible codes are
```
bandAlpha1 = [
    [1.36144854e-05],
    [1.01637179e-05],
    [1.18642372e-05],
    [1.41590759e-05],
    [1.52691734e-05],
    [1.33787913e-05],
    [1.77864235e-05],
    [1.44647350e-05],
    [2.74915514e-05],
    [3.23384006e-05],
    [2.01845999e-05],
    [7.55948936e-06],
    [9.03805553e-06],
    [5.27009746e-06],
    [8.32830484e-06],
    [8.43511259e-06],
    [1.64113800e-05],
    [1.82010771e-05],
    [2.04712873e-05],
    [6.43561709e-06]]
bandHighBeta = [
    [9.48504150e-05],
    [5.74889957e-05],
    [9.19252640e-05],
    [8.93399030e-05],
    [4.18493759e-05],
    [3.26190943e-05],
    [2.76993552e-05],
    [1.18445736e-05],
    [1.24432218e-05],
    [2.84315098e-05],
    [7.61318171e-05],
    [4.71576660e-05],
    [3.18943913e-05],
    [2.00989813e-05],
    [1.72555372e-05],
    [2.70839355e-05],
    [5.72236646e-05],
    [3.24518943e-05],
    [1.88565039e-05],
    [3.15454230e-05]]
bandGamma = [
    [1.25188644e-06],
    [1.11386832e-06],
    [8.60886848e-07],
    [8.30455560e-07],
    [6.95102232e-07],
    [5.05349294e-07],
    [5.19661342e-07],
    [4.49833399e-07],
    [5.48951741e-07],
    [6.86494014e-07],
    [8.64004756e-07],
    [1.54017029e-06],
    [3.24345057e-06],
    [1.81364017e-06],
    [6.71151020e-07],
    [8.67693066e-07],
    [9.21449764e-07],
    [7.49581188e-07],
    [5.50885141e-07],
    [4.22944226e-07]]

import mne
import matplotlib.pyplot as plt
import numpy as np
freBandNames = ['bandAlpha1', 'bandHighBeta', 'bandGamma']

import matplotlib
print(matplotlib.__version__)
#print(bandDelta+bandHighBeta+bandGamma)
voltageFactor = 1000000
toposcalar = [ min(bandAlpha1+bandHighBeta+bandGamma)[0]*voltageFactor, max(bandAlpha1+bandHighBeta+bandGamma)[0]*voltageFactor ] #calculate topomap scalar
print(toposcalar[0],toposcalar[1])

ch_names_1020sys = ['Fp1', 'Fp2','F3','F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'A2']
info = mne.create_info(ch_names=ch_names_1020sys, sfreq=256, ch_types='eeg')

evoked = []
evoked.append( mne.EvokedArray(bandAlpha1, info) )
evoked.append( mne.EvokedArray(bandHighBeta, info) )
evoked.append( mne.EvokedArray(bandGamma, info) )

ten_twenty_montage = mne.channels.make_standard_montage('standard_1020')

fig, ax = plt.subplots(1, 6, gridspec_kw=dict(width_ratios= 3*[12, 1]))

for i in range(len(evoked)):
    evoked[i].set_montage(ten_twenty_montage)

    evoked[i].plot_topomap(times=0, vmin = 0, vmax=toposcalar[1], time_unit='s',time_format=None, 
                                axes = ax[i*2:i*2+2], cmap='Spectral_r', colorbar = True, units = '$\mu V^2$', show=False)
    #evoked[i].plot_topomap(times=0, time_unit='s',time_format=None, 
    #                        axes = ax[i*2:i*2+2], cmap='Spectral_r', colorbar = True, units = '$\mu V^2$', show=False)
    
    # set title for this topomap
    ax[i*2].set_title(str(freBandNames[i]), fontsize=10)

# adjust subplot position
fig.subplots_adjust(top=0.9,bottom=0.1)
# give the figure a main title
plt.suptitle('Power Comparison')

evoked[2].plot_topomap(times=0, vmin=0, time_unit='s',time_format=None, cmap='Spectral_r', colorbar = True, units = '$\mu V^2$', show=False)
plt.show()
```

datalw, 2020-07-07:
No ones know why this is happenining? It also happens if I only show one colorbar using a single plot_topomap function.
[image]

larsoner, 2020-07-08:
I can reproduce. @drammock it's possible this is related to the recent colorbar refactoring, can you take a look?

drammock, 2020-07-08:
Turns out not to be related to the recent refactor. The reason this is happening is that, if contour lines are drawn, the colorbar tick marks are set to match the actual contour line values that are plotted. 

https://github.com/mne-tools/mne-python/blob/2965cc3644538c97b297f131bbb46ce9f954ea6f/mne/viz/topomap.py#L1763-L1764

Since your third plot (gamma band) has very low variance compared to the other two (and hence has only one contour line), it ends up with only one tick.

There is also an unrelated problem that the colorbars are wrongly getting set to be draggable/interactive, but that's not the cause of the tick problem.

I'll open a PR with a fix shortly.

PR #7980 (2020-07-09): VIZ, BUG: fix tickmarks in evoked topomap colorbar
closes #7956 

When the last topomap drawn has low variance and hence no visible contour lines, only one tickmark is drawn on the colorbar, because it uses the *plotted* contours instead of the precomputed contour levels based on vmin/vmax.

https://github.com/mne-tools/mne-python/blob/2965cc3644538c97b297f131bbb46ce9f954ea6f/mne/viz/topomap.py#L1763-L1764
closed 2020-07-09 by larsoner