mne-tools/mne-python #7958: "reject_tmin" and "reject_tmax" params in Epochs are not applied for "reject_by_annotation"

sappelhoff, 2020-07-03:
first reported [on gitter](https://gitter.im/mne-tools/mne-python?at=5efdb0046c06cd1bf469ce70)

#### Describe the bug
The `reject_tmin` and `reject_tmax` parameters from [`Epochs`](https://mne.tools/dev/generated/mne.Epochs.html?highlight=epochs#mne.Epochs) are not used for the `reject_by_annotation` parameter. But instead only for the `reject` and `flat` parameters.

Look at [this part in the code](https://github.com/mne-tools/mne-python/blob/517c49612da33491fe4f147588deb00072526874/mne/epochs.py#L2130-L2135).

#### Steps to reproduce

MWE:

```Python
# Create raw data
sfreq = 1000
times = np.arange(0, 10, 0.001)
sin = np.sin(times * 10)
cos = np.cos(times * 10)
data = np.array([sin, cos])
ch_types = ['mag', 'grad']
ch_names = ['sin', 'cos']
info = mne.create_info(ch_names=ch_names, sfreq=sfreq, ch_types=ch_types)
raw = mne.io.RawArray(data, info)

# We have an event at 2 seconds
events = np.array([[2000, 0, 1]])

# There is a bad segment from 1 to 1.5 seconds
onset=[1]
duration=[0.5]
description=['BAD']
annots = mne.Annotations(onset, duration, description)

raw.set_annotations(annots)

# Create epochs from 1s to 3s, centered on the event at 2s.
# this epoch will be rejected_by_annotation,  because there is an overlap
# with a bad segment (1s to 1.5s)
epochs = mne.Epochs(raw, events, tmin=-1, tmax=1, preload=True, reject_by_annotation=True)
assert len(epochs) == 0

# Now setting `reject_tmin` so that there is NO overlap ... should not reject
# any epochs ... because bad segment is from 1s to 1.5s ... but epochs "reject" window
# is from 1.8s to 3s
epochs = mne.Epochs(raw, events, tmin=-1, tmax=1, reject_tmin=-0.2, preload=True, reject_by_annotation=True)

# This errors
assert len(epochs) == 1, "oups!"
```

#### Expected results
`reject_tmin` and `reject_tmax` should have an effect on how `reject_by_annotation` is applied.

Example: When an epoch from -2s to 3s relative to 0 was formed with `reject_tmin=-0.2`, then that epoch MUST NOT be rejected if there is an overlap with a `BAD` annotation from, say -2s to -1.5s

larsoner, 2020-07-06:
Agreed, @sappelhoff do you have time to look into it?

sappelhoff, 2020-07-06:
yes, I'll look into it!

PR #7967 (2020-07-07): [MRG] Fix reject_tmin and reject_tmax for reject_by_annotation in mne.Epochs
closes #7958
closed 2020-07-07 by agramfort