mne-tools/mne-python #7959: Feat Requests: pyvista - Brain/timeviewer

bloyl, 2020-07-03:
The pyvista, stc plotting is great and already really useful for reviewing data.

here are a few thoughts (in no particular order) I've had using it as part of an analysis pipeline for the last week.

1) there is no `brain.remove_labels()` which is mentioned in the `add_label` docs.
2) It would be useful to return vertices that are interactively added using `timeviewer`. This would let you save these points and later use them for analysis.
3) is it possible to link two `timeviewer`s so they share the same point picking and optionally the same camera/view. I wouldn't want them time synced (at least not by default) so you can look at different types of stcs.
4) I wonder what the opportunities are for adding labels and having time_viewer display their wave forms. Exposing the parameters for `extract_label_time_course` via the gui might be a pain though.

GuillaumeFavelier, 2020-07-06:
> The pyvista, stc plotting is great and already really useful for reviewing data.

Thanks for the nice feedback!

1) This feature was not on the top of the list but this issue can definitely increase the priority 
2) Good idea! We could create a function `get_picked_points()` or `get_picked_vertices()`, what do you say?
3) Only the time controllers can be linked at the moment (colorbar in the todo list). I can see the camera controls being shared easily and picking could be tricky (but possible!).
4) @agramfort  what do you think about this one?

I'll update #7162 with your comments and link the PRs to this issue too if you want :+1:

agramfort, 2020-07-06:
4. is what mne_analyze does historically so I think there is indeed a user
base for this. I can do you a demo

>

GuillaumeFavelier, 2020-09-09:
I wanted to work on 4) but I realized that `extract_label_time_course()` needs the `src` parameter (source spaces for left and right hemisphere) which is not available with `_Brain`. How can I fix this?

I had [this example](https://mne.tools/dev/auto_examples/visualization/plot_parcellation.html?highlight=plot_parcellation) in mind.

agramfort, 2020-09-09:
you should be able to extend extract_label_time_course allowing to
pass white surfaces instead of src.
for extract_label_time_course you need the locations and normals of
the vertices present in the stc.

larsoner, 2020-09-09:
> you should be able to extend extract_label_time_course allowing to
pass white surfaces instead of src.
> for extract_label_time_course you need the locations and normals of
the vertices present in the stc.

Unfortunately this will be quite slow since you should compute the normal directions using patch info. Better I think to allow people to pass `src` to `stc.plot`. It's not much to ask, you need it for `plot_volume_source_estimates` anyway.

agramfort, 2020-09-09:
ok fair enough

>

closed 2020-12-17 by larsoner