mne-tools/mne-python #7966: Value Error Read Raw EGI Files

SaadEphlux, 2020-07-06:
Hi all, I am trying to read the raw egi file can anyone help me with this error. 

[image]

larsoner, 2020-07-06:
Can you upload the file somewhere so we can look? Do you get this error if you use an EGI reader using the official EGI software?

SaadEphlux, 2020-07-06:
I will have to ask my team if I can share the file or not since its not mine. We dont have NETSTATION available to us now and due to COVID its difficult to get it. Is there any other EGI reader?

cbrnr, 2020-07-06:
You could try http://biosig.sourceforge.net/ - the main toolbox is for MATLAB, but there is also a Python port available. I think BioSig can read EGI: http://pub.ist.ac.at/~schloegl/biosig/TESTED

SaadEphlux, 2020-07-07:
Thanks for the suggestion I will try this out. Any other suggestions?