mne-tools/mne-python #7973: FR: REST method for EEG re-referencing

agramfort, 2020-07-07:
fieldtrip implements the REST method for EEG data re-referecing. See:

https://github.com/fieldtrip/fieldtrip/blob/master/preproc/ft_preproc_rereference.m#L140

it's actually quite simple.

EEG folks : @cbrnr @hoechenberger @sappelhoff ... do you see an interest in this?

sappelhoff, 2020-07-07:
I think it's interesting and I'd certainly like to try it on my data if it were available in MNE. But I don't feel an urge to implement it myself :)

hoechenberger, 2020-07-07:
> fieldtrip implements the REST method for EEG data re-referecing. See:
> 
> https://github.com/fieldtrip/fieldtrip/blob/master/preproc/ft_preproc_rereference.m#L140
> 
> it's actually quite simple.
> 
> EEG folks : @cbrnr @hoechenberger @sappelhoff ... do you see an interest in this?

I read about this method recently, but I was not convinced that it would be worth the effort to implement and use it. But of course, once we do have an implementation, we could run some quantitative comparisons :)

Would certainly be nice to have it in MNE! What would the API look like? `mne.set_eeg_reference('REST')`?

agramfort, 2020-07-07:
yes

mne.set_eeg_reference('REST', fwd=fwd)

>

cbrnr, 2020-07-07:
I'm probably not going to use it anytime soon because the additional requirements (lead field matrix) do not seem to yield strikingly improved results. But if anyone wants to implement it I won't object of course. Note that a Python implementation already exists here: https://github.com/HaoZ94/restpython

PR #8132 (2020-08-24): MRG, ENH: Add REST
Closes #7973

@sappelhoff can you try it and see if it looks correct? Based on the math I think it should be, but based on the output it looks quite different from other referencing schemes.

If it's incorrect, it would be great if we could get a gain matrix and REST-ed data from FieldTrip or some other software with a working implementation, and compare to their results.

Todo:

- [x] manually test against FieldTrip
closed 2020-08-24 by hoechenberger