mne-tools/mne-python #7975: to_data_frame() method for EpochsTFR objects

eioe, 2020-07-07:
Are there plans to implement a `to_data_frame()` method for `EpochsTFR` and `AverageTFR` objects?

I would like to export/save the according data for further analysis in R. 
I just recently learned about the `to_data_frame()` methods for `Epoch` and `Evoked` objects and find them terrifically useful. Would be great to get the same luxury for tfr data. 

I'm also thanksful for hints regarding major obstacles that speak against it or if there are already similar implementations that I'm not aware of. 
Thanks!

larsoner, 2020-07-07:
I think it would be fine to implement this for the TFR objects, do you want to give it a shot?

eioe, 2020-07-07:
I will see what I achieve. However, I'm still quite new to MNE and have little experience with PRs into professionnally mainatained projects. So this will take a while and it's likely that I will come back for help. 

If someone else feels like implementing this earlier, please go for it. :)

larsoner, 2020-07-07:
> However, I'm still quite new to MNE and have little experience with PRs into professionnally mainatained projects. So this will take a while and it's likely that I will come back for help.

No problem! We're here to help people work through the contributing process, so feel free to take a stab at an initial implementation, open a pull request, and we can iterate from there. It can help to read the [contributing guide](https://mne.tools/dev/install/contributing.html#overview-of-contribution-process) to help get oriented with the overall process.

eioe, 2020-07-07:
Thanks! Yes, I worked through the guide and set up a local dev enviroment. 
I'll come back when I hit a wall.

dengemann, 2020-07-07:
Sounds cool! Keep me in the loop. We can then add simultaneously a related vignette to mne-r: https://mne.tools/mne-r

eioe, 2021-03-16:
@drammock & @dengemann 
as discussed earlier, I would like to use the sprint to finally close this issue. 
Can you add it to the Sprint Program please and assign me? 
Btw the same was suggested earlier in #6619.

agramfort, 2021-03-16:
go for it

>

dengemann, 2021-03-16:
done @eioe

drammock, 2021-04-15:
closed by #9124

closed 2021-04-15 by drammock