mne-tools/mne-python #7978: DOC: order of operation in TFR functions needs to be clarified

sappelhoff, 2020-07-08:
issue came up here: https://gitter.im/mne-tools/mne-python?at=5f05de658342f4627400eee5

the tfr_morlet for time frequency decomposition has a parameter average which says:

>    If True average across Epochs.

The docs need to be updated to make it clear what the order of operations is. First averaging then decomposition? Or the other way around? What are the steps?

bloyl, 2020-07-08:
I'm relatively confident that the order is 
1) decomposition
2) averaging

if you do the following
```
freqs = np.arange(8, 31)
n_cycles = 7 
a = mne.time_frequency.tfr_morlet(epochs, freqs, n_cycles, average=True)
b = mne.time_frequency.tfr_morlet(epochs, freqs, n_cycles, average=False)
```
I'm pretty sure that `a == b.average()`

sappelhoff, 2020-07-08:
That makes me wonder why there is an `average` parameter at all --> [mne.Epochs](https://mne.tools/dev/generated/mne.Epochs.html) doesn't have one either? :thinking:

agramfort, 2020-07-08:
the average avoids storing all the TFRs in memory as the averaging is done
online.

>

bloyl, 2020-07-08:
I think there might be a problem in the test

The following line in the test should fail
https://github.com/mne-tools/mne-python/blob/2965cc3644538c97b297f131bbb46ce9f954ea6f/mne/time_frequency/tests/test_tfr.py#L109
but it passes because `assert_array_almost_equal` only checks 6 decimal places and `(Pdb) power.data.max()
2.1955190395488294e-21`

and in fact `assert_array_almost_equal(power.data, power_evoked.data, 22)` fails with a `Max relative difference: 681137.62395224`

larsoner, 2020-07-08:
Yeah that test looks broken. We should move to `assert_allclose` and probably just use `rtol` there (unless we expect some near-zero values, in which case a suitable `atol` should be chosen based on what "zero-like" means for the data)

agramfort, 2020-07-08:
this test makes no sense. You should not expect the TFR of the epochs to be
the same as the evoked.

>

larsoner, 2020-07-08:
> I'm relatively confident that the order is
>
> 1.decomposition
> 2. averaging

The critical part is where is the magnitude taken. decomp/mag/ave will be different from decomp/ave/mag (with the latter being equivalent to ave/decomp/mag).

> this test makes no sense. You should not expect the TFR of the epochs to be the same as the evoked.

You should expect the TFR averaged across epochs to be the same as the TFR of the evoked if the magnitude is taken at the end (or not at all) for both. Maybe that's what the test was trying to check? And maybe it was wrong about when the average is taken (i.e., before or after the magnitude is taken)? Someone will need to dig into the code to check...

agramfort, 2020-07-08:
for me code is correct but test is broken.

bloyl, 2020-07-08:
I agree with @agramfort and think the test was in error. the procedure to compute total power is to decomp/mag/ave. otherwise you are computing evoked total power which isn't (generally) what you want.

bloyl, 2020-07-08:
I should note that the surrounding `assert_array_almost_equal` are testing meaningful things and currently are trivially passing because of the lack of a `decimal` parameter

larsoner, 2020-07-08:
Sounds reasonable to me, @bloyl would you be up for fixing the code to use `assert_allclose` with `rtol` only and document the order of transformations in some TFR computing function (and maybe refer other ones to look at the Notes for that one)?

bloyl, 2020-07-08:
Sure although Docs aren't exactly my strong suit.

@sappelhoff what do you think of
```
    average : bool, default True
        If False return an EpochsTFR containing separate tfrs for each epoch.
        If True return an AverageTFR containing the average of all tfrs across epochs.
```

sappelhoff, 2020-07-09:
> @sappelhoff what do you think of

I think that's good but I would add some more detail:

```
    average : bool, default True
        If False return an EpochsTFR containing separate tfrs for each epoch.
        If True return an AverageTFR containing the average of all tfrs across epochs.
        AverageTFR is equivalent to EpochsTFR.average(), but setting average=True will
        be more memory efficient than setting average=False and later calling
        EpochsTFR.average().
```

PR #7998 (2020-07-13): WIP: TFR Doc/test changes
Closes #7978

@sappelhoff @larsoner @agramfort 

I have currently commented out the tests for the complex return type as they don't make sense and thus shouldn't pass.
I could remove them and just leave them as smoke tests or if anyone has a suggestion on what to regression test to I could do that. Just let me know what you think is best.

@drammock I'll admit that I don't know how all the doc linking stuff works so please let me know what needs to be changed.
closed 2020-07-13 by larsoner