mne-tools/mne-python #7981: RuntimeError: info channel name inconsistency detected, please notify mne-python developers

birajstha, 2020-07-09:
Hello
I am trying to make a topographic plot for EEG Motor imagery dataset from physionet. I am getting this error

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-18-65a7c7ac8197> in <module>
----> 1 Data.set_montage(montage)

<decorator-gen-20> in set_montage(self, montage, raise_if_subset, match_case, verbose)

~/anaconda3/lib/python3.7/site-packages/mne/io/meas_info.py in set_montage(self, montage, raise_if_subset, match_case, verbose)
    188         from ..channels.montage import _set_montage
    189         info = self if isinstance(self, Info) else self.info
--> 190         _set_montage(info, montage, raise_if_subset, match_case)
    191         return self
    192 

~/anaconda3/lib/python3.7/site-packages/mne/channels/montage.py in _set_montage(info, montage, raise_if_subset, match_case)
    705             pick_types, exclude=[], eeg=True, seeg=True, ecog=True, meg=False,
    706         )
--> 707         info_names = [info['ch_names'][ii] for ii in _pick_chs(info)]
    708         dig_names = mnt_head._get_dig_names()
    709         ref_names = [None, 'EEG000', 'REF']

~/anaconda3/lib/python3.7/site-packages/mne/io/pick.py in pick_types(info, meg, eeg, stim, eog, ecg, emg, ref_meg, misc, resp, chpi, exci, ias, syst, seeg, dipole, gof, bio, ecog, fnirs, csd, include, exclude, selection)
    381     # NOTE: Changes to this function's signature should also be changed in
    382     # PickChannelsMixin
--> 383     exclude = _check_info_exclude(info, exclude)
    384     nchan = info['nchan']
    385     pick = np.zeros(nchan, dtype=np.bool)

~/anaconda3/lib/python3.7/site-packages/mne/io/pick.py in _check_info_exclude(info, exclude)
    293 def _check_info_exclude(info, exclude):
    294     _validate_type(info, "info")
--> 295     info._check_consistency()
    296     if exclude is None:
    297         raise ValueError('exclude must be a list of strings or "bads"')

~/anaconda3/lib/python3.7/site-packages/mne/io/meas_info.py in _check_consistency(self, prepend_error)
    725             raise RuntimeError('%sinfo channel name inconsistency detected, '
    726                                'please notify mne-python developers'
--> 727                                % (prepend_error,))
    728 
    729         # make sure we have the proper datatypes

RuntimeError: info channel name inconsistency detected, please notify mne-python developers

################################################################
###################################################################
###################################################################
####################################################################
###################################################################

my code
MI_viz.ipynb

[image]
[image]
[image]
[image]
[image]
[image]
[image]

birajstha, 2020-07-09:
https://github.com/birajstha/Motor_imagery/blob/master/MI_viz.ipynb

agramfort, 2020-07-09:
did you do :

Data.info['ch_names'] = [chn.strip('.') for chn in Data.info['ch_names']]

you should not change manually the info

you can do this with rename_channels eg:

https://mne.tools/dev/generated/mne.io.BaseRaw.html#mne.io.BaseRaw.rename_channels

birajstha, 2020-07-09:
Thank you so much!! It worked 

I used 
Data.rename_channels(lambda x: x.strip('.'))

then,  Data.rename_channels(ch_map) to rename the remaining channels.

closed 2020-07-09 by birajstha