mne-tools/mne-python #7984: Average over channels

HanBnrd, 2020-07-09:
Hello!

(This issue has been raised in the mne_analysis mailing list)

#### Describe the problem
There is currently no way to average signals over channels on a Raw, Epochs or Evoked object easily. One can average signals over channels with the "combine" parameter in some visualisation options but it is not possible to modify a Raw, Epochs or Evoked instance by performing an average over channels.

#### Describe your solution
One solution could be to add a new method to Raw, Epochs and Evoked enabling to combine channels (with mean or median options for example). It could for example take as one of the parameters a dictionary where each ROI (new pseudo-channels) would be associated with a list of channels to perform the combination on. This method could also eventually return the standard deviations in order to compute confidence intervals...

#### Describe possible alternatives
- One alternative is to use for example the [plot_image](https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.plot_image) method with the "combine" parameter, but it only enables to plot signals and does not return a modified instance of Epochs.
- Another alternative is to extract data with [get_data](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.get_data), perform the operation with [numpy.mean](https://numpy.org/doc/stable/reference/generated/numpy.mean.html), recreate channel information with pseudo-channels thanks to [create_info](https://mne.tools/stable/generated/mne.create_info.html) and recreate a Raw object from a numpy array with [RawArray](https://mne.tools/stable/generated/mne.io.RawArray.html).

#### Additional context
No additional context.

Please let me know if that makes sense?

Many thanks!
Johann

rob-luke, 2020-07-11:
I would use this feature if it existed. If you are going to create this I would have the function take a dictionary similar to plot image... `group_by=dict(Left_ROI=[1, 2, 3, 4], Right_ROI=[5, 6, 7, 8])`

HanBnrd, 2020-07-11:
Good idea thank you @rob-luke.
Do you think it would be better to have this doing only an average (in which case we could have an "average_channels" method) or is it better to also have the possibility to get the median instead (in which case a "combine_channels" method would be more appropriate I guess)?

agramfort, 2020-07-11:
Be careful that you then need to discard channel locations for an average channel

drammock, 2020-07-11:
my two cents:

- `combine_channels(..., method='mean')` with default `mean` and ability to take a few keyword strings (mean, median, std) that resolve to numpy functions, or a callable.
- I'd prefer an API more like `combine_channels(left_roi=(1, 2, 3), right_roi=(4, 5, 6), method='median')` rather than having to wrap the ROIs into a dict (think of how `dict.update()` works). Not sure if that presents difficulties in arg parsing or documentation though...
- I'm assuming any channels not provided in the ROIs are simply dropped?  Seems like they should be, though maybe there are use cases I'm not aware of where you'd want a mix of original channels and combined channels?

HanBnrd, 2020-07-11:
Thanks @drammock.

- I like the idea of a combine_channels method which defaults to mean.

- Not sure what you mean with the `dict.update()`, could you please give more details? I think a dict can let more freedom to users as they could create as many ROIs as they want, but also give them custom names ('left_roi' and 'right_roi' being possibilities). I feel like it could become very complex with a lot of ROIs without a dict... I was thinking of it the same way one can use an [evoked_dict](https://mne.tools/stable/auto_tutorials/preprocessing/plot_70_fnirs_processing.html?#plot-standard-fnirs-response-image) if that makes sense.

- Dropping channels not provided would make sense. With a dict it would be possible to specify a ROI as being only one channel if someone wants to keep original channels. Maybe in some cases users may want to keep for example an EEG reference channel as is. I'm working mainly with fNIRS at the moment and ROIs make probably more sense with it as we often don't have proper locations (with eg. 10-20 system).

drammock, 2020-07-12:
>     * Not sure what you mean with the `dict.update()`, could you please give more details?

`dict.update()` allows but doesn't require a `dict` argument; you can pass the new key/value pairs as parameters.  Here's an example:

```python
>>> foo = dict(a=1)
>>> foo
{'a': 1}
>>> foo.update(b=2, c=3, d=4)  # can pass each new key/value as a parameter
>>> foo
{'a': 1, 'b': 2, 'c': 3, 'd': 4}
>>> foo.update(dict(e=5, f=6))  # can pass a dict
>>> foo
{'a': 1, 'b': 2, 'c': 3, 'd': 4, 'e': 5, 'f': 6}
```

My example with `left_roi` and `right_roi` was not meant to suggest that those two (and only those two) ROI names should be possible. They were just examples.

drammock, 2020-07-12:
>     * it would be possible to specify a ROI as being only one channel if someone wants to keep original channels.

I'm still not 100% clear that we should support this use case, but it seems like if we want to allow a mix of single channels and averaged ROI "channels", then a boolean `drop=False` should be sufficient, and the function should leave any channels not in the ROIs untouched.  Under your proposal, if a user wanted to keep say 20 single channels, they would need 20 extra dict keys or would need to pass 20 extra parameters.

To anticipate the worry of "what if they want to keep 20 single channels, create ROIs out of another 20, and drop all the rest?"  In that case, they can use existing functionality to do the dropping step first before creating the ROIs.

HanBnrd, 2020-07-12:
Alright, good point. Thanks for the explanation.
I guess I can start working on this, and I'll see how it goes. It would be my first contribution to MNE but I'm happy to do it as I think it will be a good learning process, if you're okay with it?

drammock, 2020-07-12:
Yes! We are very welcoming of new contributors, and will happily provide reviews and guidance on your first pull request :)

PR #8014 (2020-07-24): MRG, ENH: Combine channels
#### Reference issue
Closes #7984

#### What does this implement/fix?
Implementation of a way to combine channels from Raw, Epochs, or Evoked. Channels can be combined with a callable as argument or with "mean", "median", or "std".

#### Additional information
- It currently returns a new instance and does not modify the original one
- It is currently a method in `mne/channels/channels.py`, open to suggestions for a better location
- Was wondering if this sounds fine before implementing tests
closed 2020-07-24 by drammock