mne-tools/mne-python #7993: plot_projs_topomap() of data with EEG projection creates "EEG-style" topoplots even for MEG data

hoechenberger, 2020-07-10:
#### Describe the bug
Plotting SSP projectors via `Raw.plot_projs_topomap()` we noticed that in some situations, the visualized MEG fields didn't look quite right, i.e., looked like circular EEG topomaps instead like the expected "helmet-shaped" topomaps. Turns out, this only happened when among `info['projs']` was also an EEG average reference projector.

#### Steps to reproduce
```python
import os.path as op
import mne
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname, preload=True)

raw.copy().set_eeg_reference(projection=True).plot_projs_topomap()  # Add EEG avg ref projector
raw.copy().set_eeg_reference(projection=False).plot_projs_topomap()  # No EEG avg ref projector
```

#### Expected results
MEG topomap fields should follow the MEG helmet, and EEG fields should be circular.

#### Actual results
**With EEG avg ref projector:**
[image]

**Without EEG avg ref projector:**
[image]

cc @agramfort

larsoner, 2020-08-10:
@hoechenberger do you want to look into how to fix this? I can confirm the bug

hoechenberger, 2020-08-10:
I might have time for this tomorrow, sure. But if you're asking if I would also be okay with you going ahead: sure ;)

PR #8200 (2020-09-02): MRG, FIX: ch_type in plot_projs_topomap
Closes #7993

This bug was in this release so no need for `latest.inc` update. You can see the problem on `master` [here](https://mne.tools/dev/auto_tutorials/preprocessing/plot_50_artifact_correction_ssp.html#repairing-ecg-artifacts-with-ssp):

[image]
closed 2020-09-02 by larsoner