mne-tools/mne-python #7999: API: mne.preprocessing.mark_flat --> mne.preprocessing.annotate_flat ?

sappelhoff, 2020-07-11:
With the *relatively* new Annotations in MNE, there are now (at least) two functions that work on raw data and return annotations:

- [mne.preprocessing.annotate_muscle_z_score](https://mne.tools/dev/generated/mne.preprocessing.annotate_muscle_zscore.html?highlight=annotate#mne.preprocessing.annotate_muscle_zscore)
- [mne.preprocessing.annotate_movement](https://mne.tools/dev/generated/mne.preprocessing.annotate_movement.html?highlight=annotate#mne.preprocessing.annotate_movement)

There is an older function operating in a similar style: [`mne.preprocessing.mark_flat`](https://mne.tools/stable/generated/mne.preprocessing.mark_flat.html#mne.preprocessing.mark_flat)

Would it make sense to change this to `mne.preprocessing.anotate_flat` to have a set of functions with a consistent API working on raw data and providing annotations?

agramfort, 2020-07-11:
+100

>

larsoner, 2020-07-11:
That function might mark bad channels and it might annotate, depending on the data and what percentage of time a given channel is flat. So maybe annotate_mark_flat?

drammock, 2020-07-11:
> So maybe annotate_mark_flat?

this is in fact the only function with `mark_` in its name (although there is the MNE-C function mne_mark_bad_channels). So unclear to me that we need the longer, more complicated name.  It doesn't seem conceptually weird to me that a temporal-span annotation function might "give up" and just mark the whole channel in some cases. +1 for `annotate_flat`

hoechenberger, 2020-07-12:
> +1 for `annotate_flat`

+1

hoechenberger, 2020-07-12:
I just realized all of these functions return Annotations, but do not add them to the passed Raw instance. To me, this behavior seems to kind of contradict the function name? i.e. when something is called "annotate", I'd expect it to actually annotate… 🧐

cbrnr, 2020-07-12:
I know MNE methods should generally work in-place, but in the case of artifact processing I'd want an option to double-check the automatically generated annotations/bad channels. Therefore, I'd prefer such methods to return annotations and a list of bad channels.

hoechenberger, 2020-07-12:
> but in the case of artifact processing I'd want an option to double-check the automatically generated annotations/bad channels. Therefore, I'd prefer such methods to return annotations and a list of bad channels.

… which is already the case for `annotate_muscle_z_score` and `annotate_movement`; and if we rename `mark_flat` anyway, we could also change its behavior accordingly.

Still, my issue remains: if I say "annotate something", I expect "something" to be annotated. Just like when I call `list.remove()`, I expect something to be removed from the list, without requiring an additional step from my end. – If, however, the function were called e.g. `gen_movement_annotations`, I would expect it to return annotations (and not necessarily work in-place).

cbrnr, 2020-07-12:
What about `str.upper()` then?

hoechenberger, 2020-07-12:
> What about `str.upper()` then?

"upper" is not a verb, so to me the behavior is unclear unless I check the docs.

hoechenberger, 2020-07-12:
(besides, strings are immutable; so the comparison is a little off anyway… but this is an implementation detail)

cbrnr, 2020-07-12:
You could interpret it as a verb though - uppercase that string for me. And yes, strings are immutable, but I wanted to point out that even in the built-in types this is not consistent.

Regarding your expectation, if you annotate something, you could also get back the annotated object (instead of modifying in-place).

hoechenberger, 2020-07-12:
> Regarding your expectation, if you annotate something, you could also get back the annotated object

Yes, in addition to the in-place modification. Otherwise it would be, "generate an annotation please" and not "please annotate".

hoechenberger, 2020-07-12:
(I've edited my answer above)

hoechenberger, 2020-07-12:
In [compute_proj_ecg](https://mne.tools/stable/generated/mne.preprocessing.compute_proj_ecg.html#mne.preprocessing.compute_proj_ecg) and [compute_proj_eog](https://mne.tools/stable/generated/mne.preprocessing.compute_proj_eog.html#mne.preprocessing.compute_proj_eog)
we  do as the function name indicates: compute the projs, and return them. The verb, here, makes total sense. OTOH, annotate_*, without actually annotating, does not make sense to me.

drammock, 2020-07-12:
> I just realized all of these functions return Annotations, but do not add them to the passed Raw instance. To me, this behavior seems to kind of contradict the function name? i.e. when something is called "annotate", I'd expect it to actually annotate

@hoechenberger what is the concrete proposal here?  Is it to rename the existing `annotate_*` functions to `make_*_annotations`, or to keep their names and change their behavior?  If it's the latter, please open a separate issue, and we can (I think?) move forward here with renaming `mark_flat` to `annotate_flat`.

If your proposal is to keep the behavior and change the names, then we should probably do here it all at once.  But that said, I'm not bothered by the `annotate_*` naming. To me "annotate" can mean either "make notes" or "attach notes". If you think about a literary work, annotating it does not necessarily mean writing in the margins of a physical book; a separate document with page and paragraph numbers attached to each note still counts as annotation IMO. That seems quite analogous to our `Annotations` object (which incidentally, a user might simply want to save to disk in tabular format --- returning the modified `Raw` is not helpful in that case).

hoechenberger, 2020-07-12:
@drammock 
> @hoechenberger what is the concrete proposal here? Is it to rename the existing `annotate_*` functions to `make_*_annotations`, or to keep their names and change their behavior?

Since my interpretation of "to annotate" seems to be a bit of an edge case, I'm okay with leaving these functions as is (I very much **would** prefer to rename them to `make_*_annotations`, personally.)

But we should make sure that the current `mark_flat`, if renamed to `annotate_flat`, matches the behavior of the existing `annotate_*` functions, namely, stop modifying the input in-place, and return Annotations instead.

drammock, 2020-07-12:
> But we should make sure that the current `mark_flat`, if renamed to `annotate_flat`, matches the behavior of the existing `annotate_*` functions, namely, stop modifying the input in-place, and return Annotations instead.

Agreed; I think that was part of @sappelhoff's original proposal.

larsoner, 2020-07-12:
Sounds reasonable to me

larsoner, 2020-07-12:
(if someone calls `mark_flat` in 0.21 it should do the same thing as in 0.20 but give dep warning, `annotate_flat` should do whatever we want `annotate_*` functions to do)

larsoner, 2020-08-24:
@sappelhoff will you be able to take a stab at this one in the next couple of weeks?

hoechenberger, 2020-08-24:
Offering my help here too 👍

sappelhoff, 2020-08-24:
> Offering my help here too +1

In that case, I'd be happy if you took over @hoechenberger :-)

agramfort, 2020-08-25:
+1 to add this to the next release

>

PR #8229 (2020-09-10): MRG, API: Move mark_flat->annotate_flat
Closes #7999
closed 2020-09-10 by agramfort