mne-tools/mne-python #8001: Add file reader for Artinis fNIRS devices

rob-luke, 2020-07-12:
This issue is about adding support for [Artinis](https://www.artinis.com) Medical Systems [fNIRS devices](https://www.artinis.com/nirs-devices) to MNE.

The first step is to collect some small (approximately 10 second) recordings to be used in unit tests. The details of how the data is collected should be documented including all hardware and software parameters. The test data should be committed to the [mne-testing-data repository](https://github.com/mne-tools/mne-testing-data). Here is an example of the [first NIRS test file that was added](https://github.com/mne-tools/mne-testing-data/pull/51) which is a good example of the minimum level of detail preferred, additional testing data was added as support was expanded to include more [hardware](https://github.com/mne-tools/mne-testing-data/pull/52) and [software](https://github.com/mne-tools/mne-testing-data/pull/56) versions. It is handy if the data can include some trigger information too.

Based on the OxyMon Leaflet - rev. 1901 the OxyMon is a continuous wave device so we can use the type `fnirs_cw_amplitude` (see https://github.com/mne-tools/mne-python/issues/7912 for discussion of different types). This is the same data type as used for the NIRX reader, so that can be used as a template. Please correct me if I am wrong here, I don't have access to a device.

A file reader will need to be created and unit tests for the reader. As an example you can view the [NIRX file reader here](https://github.com/mne-tools/mne-python/blob/master/mne/io/nirx/nirx.py) and the [tests can be viewed here](https://github.com/mne-tools/mne-python/blob/master/mne/io/nirx/tests/test_nirx.py). It is also worth looking at the [non NIRS data readers too ](https://github.com/mne-tools/mne-python/tree/master/mne/io)as they may be more similar to the Artinis format.

larsoner, 2020-07-13:
Sounds like a good plan to me

rob-luke, 2020-07-15:
Alternatively if there is an Artinis to SNIRF converter then we can just ensure the SNIRF reader is sufficient.

HanBnrd, 2021-04-16:
Just referencing a few items that may be useful:
- The Artinis montage PR may be useful for this issue: #9141
- Loading SNIRF files from Artinis leads to the following issue: https://github.com/mne-tools/mne-python/issues/9265#issuecomment-815953357

rob-luke, 2021-10-24:
Artinis have added a tutorial on their website about how to use Artinis data with MNE https://www.artinis.com/blogpost-all/2021/fnirs-analysis-toolbox-series-mne-python so combined with the changes from @HanBnrd I guess this is closed. Please reopen if the artinis provided solution is not sufficient.

closed 2021-10-24 by rob-luke