mne-tools/mne-python #8008: API: Confirm best defaults for LCMV before 0.21

larsoner, 2020-07-14:
In #7656 we set the default `make_lcmv(..., weight_norm='sqrtm')` because it's superior to the official `'unit-noise-gain'` in extensive simulation testing. But we should just double check with @britta-wstnr (busy right now AFAIK) that this is the default we should have for 0.21 before releasing.

larsoner, 2020-07-17:
Also, @wmvanvliet should we deprecate apply_dics? Do we need a apply_dics_tfr first?

wmvanvliet, 2020-07-19:
Let's deprecate when we have a viable alternative. Just so you know, I have no intention to work on an `apply_dics_tfr` function at the moment, but if someone else wants to take a stab at it, they have my blessing (and my simulations if they want)

larsoner, 2020-09-09:
@wmvanvliet @britta-wstnr are we okay here?

wmvanvliet, 2020-09-10:
I think so!

larsoner, 2020-09-14:
Okay I'll close this for now then

larsoner, 2020-09-15:
@wmvanvliet I'm trying to speed up some examples and noticed [in master](https://mne.tools/dev/auto_examples/inverse/plot_dics_source_power.html#sphx-glr-auto-examples-inverse-plot-dics-source-power-py) the DICS example gives:

[image]

[0.20 stable has](https://mne.tools/stable/auto_examples/inverse/plot_dics_source_power.html#sphx-glr-auto-examples-inverse-plot-dics-source-power-py):

[image]

Can you look to see if this is wrong? If so a `git bisect` might tell us where things went wrong.

wmvanvliet, 2020-09-15:
master certainly is wrong...

wmvanvliet, 2020-09-15:
we need either `inversion='single'` or `reduce_rank=True` to make it work again.

larsoner, 2020-09-15:
Indeed `reduce_rank=True` works, I'll add it to #8260. Thanks for the quick investigation!

PR #8260 (2020-09-15): MAINT: Speed up more [skip travis]
Lots of little memory tweaks, hopefully once green+merge CircleCI builds properly.

Closes #8008
closed 2020-09-15 by larsoner