mne-tools/mne-python #8009: MAINT, DOC: bump spyder-kernels

drammock, 2020-07-14:
Once Spyder 4.1.4 hits the anaconda `defaults` channel, we'll want to update the install instructions to state that the `base` env should have `spyder>=4.1.4`, and update our `environment.yml` to require `spyder-kernels>=1.9.2`. This is necessary because on OSX Catalina (and presumably also on Big Sur) there is an issue with spyder running under py3.7 in base being unable to set the console to use a py3.8 executable (which currently happens by default when installing with our env file).

Check here periodically to see if the relevant versions are out yet:

- https://anaconda.org/anaconda/spyder
- https://anaconda.org/anaconda/spyder-kernels  (can use pip to get 1.9.2 if the conda release lags spyder 4.1.4)

Crossref spyder-ide/spyder#13252

drammock, 2020-07-15:
One of the Spyder devs (@ccordoba12) just mentioned that they expect to release platform binaries of Spyder for Windows and macOS in September (it is already available in Linux via `apt`, though a very old version; currently 3.3.6 via apt, latest is 4.1.4). TBD whether this news actually makes our installation guidance for MNE-Python easier, depending on how tightly the platform binary integrates with the Anaconda Navigator GUI.  Here is what I think would be necessary to go back to installing MNE-Python into `base` env:

1. Anaconda stops installing Spyder into `base` by default once the platform binaries are available; OR
2. Anaconda Navigator detects if Spyder had been installed as a platform binary, and (in Navigator GUI) masks the env-specific Spyder launcher with a launcher for the binary; OR
3. We recommend that all users *uninstall* Spyder from `base` before installing MNE-Python & its dependencies, and recommend that afterward they install the Spyder platform binary if they want it; OR
4. We recommend new users install *miniconda* + Spyder platform binary (if they want it) so that Spyder isn't in `base` by default

Option 5 is to ignore the platform binaries, keep using the "modular approach" where Spyder is in `base` env and MNE-Python is in `mne` env, and tell users to set the executable in the Spyder GUI to use the version of Python in the `mne` env.

Opinions welcome.

larsoner, 2020-07-22:
Hopefully they go with (1). Seems like the easiest approach

cbrnr, 2020-07-22:
(1) would definitely be the best approach. Also, both packages have now been bumped to the versions stated at the top.

ccordoba12, 2020-07-22:
Hey guys, I also think (1) would be the easiest option not just for your project but for novice users in general. But there's one important point you should be aware of: our installers will be provided by us and (most probably) they'll use pip packages. That's because the Python installer creation tools (like py2app, py2exe and Pyinstaller) work much better with pip packages.

However, I'll talk to the Anaconda guys to see if they can stop installing Spyder into `base`.

PR #8020 (2021-01-12): MRG: bump spyder version
closes #8009

currently WIP because the fix doesn't actually work in my local testing; waiting to hear back from Spyder devs.
closed 2021-01-12 by larsoner