mne-tools/mne-python #8013: BUG: TFR.plot issues with title

sappelhoff, 2020-07-15:
I am experiencing several issues with the `.plot` method of TFR objects. I suspect that there are several things off, but for now I found this obvious bug:

These are valid inputs for the `title` parameter:

https://github.com/mne-tools/mne-python/blob/638847ea5539be32919b6ac4dd85ef8c4c3b79aa/mne/time_frequency/tfr.py#L1136-L1139

Yet in the code, we have this logic:

https://github.com/mne-tools/mne-python/blob/638847ea5539be32919b6ac4dd85ef8c4c3b79aa/mne/time_frequency/tfr.py#L1269-L1278

Hence, when I pass `title='auto'`, my figure will have the title `auto`

sappelhoff, 2020-07-15:
It appears that this is also not true:

https://github.com/mne-tools/mne-python/blob/638847ea5539be32919b6ac4dd85ef8c4c3b79aa/mne/time_frequency/tfr.py#L1183-L1185

if `combine=None` and `picks=[ch1, ch2]`, I still only get 1 figure, instead of the advertised 2 figures.

larsoner, 2020-07-15:
AFAIK these functions are not super widely used, so these are indeed likely bugs

bloyl, 2020-07-16:
yes that function is unusable for anything other than plotting a single TFR.
If you look at the code the `return` is inside the loop so it shouldn't be surprising that it doesn't work with multiple channels.
https://github.com/mne-tools/mne-python/blob/056c273f1bb96bb3229e60874122f830f946746c/mne/time_frequency/tfr.py#L1280-L1284

For what its worth it is relatively easy to write your own wrapper that will do what you need it to. I use something based off the plotting in https://mne.tools/stable/auto_examples/time_frequency/plot_time_frequency_erds.html#sphx-glr-auto-examples-time-frequency-plot-time-frequency-erds-py to plot multi channel tfrs.

sappelhoff, 2020-07-16:
Thanks @bloyl that looks good. I'll give it a try tomorrow, but also attempted a fix of the plot method in #8019. I'd be happy if you could try and review if you have time.

sappelhoff, 2020-07-17:
the function also does not plot units for the colorbar label.

sappelhoff, 2021-02-24:
not sure, but perhaps this could be an "easy" issue for the sprint? @drammock? it would involve digging into the `.plot` method, finding all bugs and fixing them (without extending functionality), and adjusting docstrings so that they accurately reflect what is happening.

eort, 2021-03-16:
I could try some digging.

sappelhoff, 2021-03-16:
That would be wonderful @eort

eort, 2021-03-16:
oh boy, this is really quite a challenge to unpack :)

 a couple of design question: 

##### The `auto` value for `title`, number of picks, and `combine` are co-dependent

 `title='auto'` is supposed to concatenate channels names for up to 6 picks. This would only make sense when there are multiple picks *and* those picks are combined into a single figure (with the `combine` keyword). Otherwise, I don't see why I would want to have all picks in the title of a figure that only shows one pick. In case multiple picks are combined, I am not sure whether listing the channels is the best representation. At least in [viz/utils.py](https://github.com/mne-tools/mne-python/blob/0a61e1c7c271829bb3cc9480ea8fb08142c91e7b/mne/viz/utils.py#L1815) it seems to be to produce a title like `RMS/AVG of X electrodes` instead of just a list of channel names. 
Perhaps a combination of the two would be better? (e.g. `AVG of Ch1, Ch2, ...`, for fewer than 6 picks).

In any case, what `mne` promises `auto` will do, is only reasonable for a specific combination of parameters. What would be the best approach to deal with that? Explain it in the doc? Raising exceptions? Silently adjusting the output whenever it does not make sense (i.e. multiple picks, uncombined)?  

##### return separate figures or subplots?

Right now, the function promises to return multiple figures (even though it actually only returns a single figure). Is the idea to have a separate figure for each pick, or would it make sense to collect the tr plots in subplots of a figure and return that one? Or allow for both? If it should be possible to return multiple figures, they probably should be returned as a list of figures. 

Tomorrow during the sprint, I'll try to move further into this rabbit hole. Happy to chat then!

sappelhoff, 2021-03-16:
> At least in viz/utils.py it seems to be to produce a title like RMS/AVG of X electrodes instead of just a list of channel names.

Agreed, I like the `RMS/AVG of X electrodes` better. The combination you propose is also fine IMHO.

> what mne promises auto will do, is only reasonable for a specific combination of parameters. What would be the best approach to deal with that? Explain it in the doc? Raising exceptions? Silently adjusting the output whenever it does not make sense (i.e. multiple picks, uncombined)?

explaining it in the docstring might be the easiest choice here. If it's properly documented there, adjusting the title silently (or with a runtime warning) can be done if the parameters pushed by the user don't make sense for `title` ... 

> Right now, the function promises to return multiple figures (even though it actually only returns a single figure). Is the idea to have a separate figure for each pick, or would it make sense to [...]

I think allowing for both would be nice, returning a list of figures would be a good solution IMHO.

> Tomorrow during the sprint, I'll try to move further into this rabbit hole. Happy to chat then!

I am glad you are doing this :-) 

maybe after the sprint these lines within the mne codebase will finally spark joy again.

PR #9150 (2021-04-26): [BUG] Tfr misc issues
#### Reference issue
Example: Fixes #8013 .

#### What does this implement/fix?
- function returns a list of figures rather than a single figure, for every pick that is being plotted
- the title is set for each figure separately, depending on the arguments of `title` and `combine` in following way:
    - if `title` is `None` or a `String`, set the title of every figure accordingly
    - if `title` is `'auto'` and `combine` is `None`, set the title of every figure to the channel name that is being plotted
    - if `title` is `'auto'` and `combine` is `not None`, set the ti [... cut]
closed 2021-04-26 by larsoner