mne-tools/mne-python #8025: BUG: montage.plot mismatch error when no data object is involved

sappelhoff, 2020-07-17:
first discussed in thie [gitter threat](https://gitter.im/mne-tools/mne-python?at=5f115f6aa874ff08d27f229d)

**Show the bug (MNE master):**

```Python
import os.path as op                                                    
import mne                                                              

datapath = mne.datasets.testing.data_path()                             
montage_fname = op.join(datapath, 'montage', 'captrak_coords.bvct')      
montage = mne.channels.read_dig_captrak(montage_fname)                  
montage.plot()                                                          
```

**Expected behavior**

... it should plot the montage

**Actual behavior**

```
ValueError                                Traceback (most recent call last)
<ipython-input-7-b9409427730c> in <module>
----> 1 montage.plot()

~/miniconda3/envs/sp/lib/python3.6/site-packages/mne/channels/montage.py in plot(self, scale_factor, show_names, kind, show, sphere)
    192         return plot_montage(self, scale_factor=scale_factor,
    193                             show_names=show_names, kind=kind, show=show,
--> 194                             sphere=sphere)
    195 
    196     def save(self, fname):

<decorator-gen-141> in plot_montage(montage, scale_factor, show_names, kind, show, sphere, verbose)

~/miniconda3/envs/sp/lib/python3.6/site-packages/mne/viz/montage.py in plot_montage(montage, scale_factor, show_names, kind, show, sphere, verbose)
     69     info = create_info(ch_names, sfreq=256, ch_types="eeg")
     70     RawArray(np.zeros((len(ch_names), 1)), info,
---> 71              copy=None).set_montage(montage)
     72     fig = plot_sensors(info, kind=kind, show_names=show_names, show=show,
     73                        title=title, sphere=sphere)

<decorator-gen-20> in set_montage(self, montage, raise_if_subset, match_case, verbose)

~/miniconda3/envs/sp/lib/python3.6/site-packages/mne/io/meas_info.py in set_montage(self, montage, raise_if_subset, match_case, verbose)
    188         from ..channels.montage import _set_montage
    189         info = self if isinstance(self, Info) else self.info
--> 190         _set_montage(info, montage, raise_if_subset, match_case)
    191         return self
    192 

~/miniconda3/envs/sp/lib/python3.6/site-packages/mne/channels/montage.py in _set_montage(info, montage, raise_if_subset, match_case)
    736                    (len(not_in_montage), _pl(not_in_montage), not_in_montage))
    737             if raise_if_subset:
--> 738                 raise ValueError(msg)
    739             else:
    740                 warn(msg)

ValueError: DigMontage is a only a subset of info. There are 1 channel position not present in the DigMontage. The required channels are: ['REF']
```

agramfort, 2020-07-18:
can you take a stab at the fix?

>

sappelhoff, 2020-07-19:
Maybe at some point, but not now --> I would say this issue is "up for grabs".

PR #8178 (2020-09-01): MRG, BUG: Fix montage plotting, deduplication, and preload check
Fix a few small bugs before release. Pooled together because the changes are all pretty small and hopefully easy to understand, so I figured I'd try to save some CI cycles. Can split back out if necessary.

Closes #8025
Closes #7859
Closes #7452
closed 2020-09-01 by larsoner