mne-tools/mne-python #8026: FR: Document units for TFR objects and adjust/track depending on baseline correction

sappelhoff, 2020-07-17:
Came up in this [gitter thread](https://gitter.im/mne-tools/mne-python?at=5f120516a28d973192df556b).

#### Describe the problem

The units for common TFR objects like [AverageTFR](https://mne.tools/stable/generated/mne.time_frequency.AverageTFR.html) are unclear.

#### Describe your solution

The units should be tracked ... and adjusted when a baseline correction is performed

#### Additional context

Other MNE functions track the units as well, and the colorbars or ylabels in plots are labelled with the units ... this is not the case for TFR objects yet.

bloyl, 2020-07-20:
you can get channel units from `tfr.info['chs']` specifically the `unit` and `unit_mul` or each channel's `dict`.

So I think you just need to figure out how to keep tract of the modifications that baseline correction has performed. A positive side effect of that would be to not allow additional `apply_baseline` calls once a tfr has been baseline corrected.

larsoner, 2020-08-20:
Also we probably need to check to make sure our units are correct in all our TFR/PSD functions. It's possible that some normalization factors could be wrong if we use plain `fft` and don't account for the normalization (not sure if we do or not)

mmagnuski, 2020-11-11:
> A positive side effect of that would be to not allow additional apply_baseline calls once a tfr has been baseline corrected.

Yes, that would be nice!

larsoner, 2024-07-15:
I'm going to remove the milestone from this rather than bumping it for a 13th (!) time, but it would be great if someone could tackle this at some point!