mne-tools/mne-python #8030: preload parameter not functional in make_fixed_length_epochs()

Macks3110, 2020-07-20:
#### Describe the bug
When running 'make_fixed_length_epochs', the preload parameter of the function is not passed to the inner function 'make_fixed_length_events', hence we are not able to preload the epochs. 

#### Steps to reproduce
"""
import mne
import os.path as op

data_path = mne.datasets.sample.data_path()

raw_fname = op.join(data_path, 'MEG/sample/sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(raw_fname, verbose='error')
epochs = mne.make_fixed_length_epochs(raw, duration=1.,
                             preload=True, verbose=None)
epochs.preload
"""

#### Expected results
We would expect epochs.preload to return True, as we passed 'preload=True' in the mne.make_fixed_length_epochs function.

#### Actual results
epochs.preload actually returns False, and so when we want to use the Epochs object in functions, it raises an error saying the data should first be loaded in memory

#### Additional information
Platform:      Darwin-18.7.0-x86_64-i386-64bit
Python:        3.7.6 (default, Jan  8 2020, 13:42:34)  [Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/Maxime/anaconda3/envs/eeg_analysis/bin/python
CPU:           i386: 4 cores
Memory:        8.0 GB
mne:           0.20.5
numpy:         1.18.1 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://backend_interagg}
sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.24.0
vtk:           8.2.0

agramfort, 2020-07-20:
this parameter is actually not necessary in make_fixed_length_epochs
function. It could be removed.
You can do what you want with

epochs.load_data()

hoechenberger, 2020-08-03:
> this parameter is actually not necessary in make_fixed_length_epochs
> function. It could be removed.
> You can do what you want with
> 
> epochs.load_data()

While I do agree that this would work, it's kind of inconsistent with how we call the `Epochs` objects constructor (i.e. how users create epochs via `mne.Epochs`): `mne.Epochs` **does** have a `preload` kwarg, and for the sake of consistency, `mne.make_fixed_length_epochs` should have one too, I believe.

hoechenberger, 2020-08-03:
Meaning that we should look into fixing this :)

agramfort, 2020-08-03:
no strong feeling. If you want to add it no objection

yh-luo, 2020-08-10:
This was fixed in #8103. `preload` was not passed into `Epochs` previously.

closed 2020-08-10 by larsoner