mne-tools/mne-python #8031: Getting error while ploting with Matplotlib 3.3.0 installed

balandongiv, 2020-07-20:
#### Describe the bug
After upgrading **Matplotlib** `3.2.2` to `3.3.0`, calling the plot() module result in the compiler return an error:

`ValueError: The number of FixedLocator locations (20), usually from a call to set_ticks, does not match the number of ticklabels (1).`

The full error thrown are:

```
  File "\lib\site-packages\mne\io\base.py", line 1378, in plot
    show_scalebars=show_scalebars, verbose=verbose)
  File "<decorator-gen-138>", line 21, in plot_raw
  File "\lib\site-packages\mne\viz\raw.py", line 415, in plot_raw
    n_channels)
  File "\lib\site-packages\mne\viz\raw.py", line 722, in _prepare_mne_browse_raw
    ax.set_yticklabels(['X' * max([len(ch) for ch in info['ch_names']])])
  File "\lib\site-packages\matplotlib\axes\_base.py", line 63, in wrapper
    return get_method(self)(*args, **kwargs)
  File "\lib\site-packages\matplotlib\cbook\deprecation.py", line 451, in wrapper
    return func(*args, **kwargs)
  File "\lib\site-packages\matplotlib\axis.py", line 1793, in _set_ticklabels
    return self.set_ticklabels(labels, minor=minor, **kwargs)
  File "lib\site-packages\matplotlib\axis.py", line 1715, in set_ticklabels
    "The number of FixedLocator locations"
ValueError: The number of FixedLocator locations (20), usually from a call to set_ticks, does not match the number of ticklabels (1).
```

Process finished with exit code 1

#### Steps to reproduce
Upgrade Matplotlib 3.3.0

agramfort, 2020-07-20:
can you share a full snippet to replicate the pb? thanks

>

larsoner, 2020-07-20:
This is probably https://github.com/mne-tools/mne-python/pull/7811, which I can backport and release.

I've found a bug where scrolling up/down can break things, though, so I'll fix that first then backport both PRs

balandongiv, 2020-07-21:
Hi @larsoner, you are right, upgrading from mne-0.20.7 to mne-0.21.dev0 does resolve the issue. Thanks for the patch
Hi @agramfort , are you still interested with the full snippet?

closed 2020-07-21 by balandongiv

agramfort, 2020-07-21:
no need we're good

>

cbrnr, 2020-09-02:
@larsoner any plans to backport this? It's still 2 weeks until 0.21, and currently `raw.plot()` doesn't work with 0.20.7 and the latest Matplotlib.