mne-tools/mne-python #8032: The plot_arrowmap does not display arrows

wsxmxm, 2020-07-20:
#### Describe the bug
I am using MNE 0.20.6-Python 3.7.6 on a Windows OS. The example code plot_evoked_arrowmap.py on  https://mne.tools/stable/auto_examples/visualization/plot_evoked_arrowmap.html does not display arrows. There is a UserWarning caused by plot_arrowmap(): Tight layout not applied. The left and right margins cannot be made large enough to accommodate all axes decorations.

#### Steps to reproduce
〈import numpy as np
from mne.datasets import sample
from mne import read_evokeds
from mne.viz import plot_arrowmap
path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'
condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))
evoked_mag = evoked.copy().pick_types(meg='mag')
max_time_idx = np.abs(evoked_mag.data).mean(axis=0).argmax()
plot_arrowmap(evoked_mag.data[:, max_time_idx], evoked_mag.info)〉

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
〈The first picture in the tutorial https://mne.tools/stable/auto_examples/visualization/plot_evoked_arrowmap.html. It is an arrowmap. 
[image]
〉

#### Actual results
〈The topomap is shown but the expected arrows are missing. And there is a UserWarning: Tight layout not applied. The left and right margins cannot be made large enough to accommodate all axes decorations.
[image]
〉

#### Additional information
〈Platform:      Windows-8.1-6.3.9600-SP0
Python:        3.7.6 (default, Jan  8 2020, 20:23:39) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\xm\anaconda3\python.exe
CPU:           Intel64 Family 6 Model 61 Stepping 4, GenuineIntel: 4 cores
Memory:        3.7 GB

mne:           0.20.6
numpy:         1.18.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.1.0
cupy:          Not found
pandas:        1.0.1
dipy:          1.1.1
mayavi:        4.7.2.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.24.0
vtk:           9.0.0〉

closed 2020-07-20 by larsoner

wsxmxm, 2020-07-21:
> I can confirm on Linux. A quick work-around is to include import
> matplotlib.pyplot as plt and plt.ion() in your script. If you do that,
> you will still get the tight_layout-related warning, but the arrows will
> get drawn.

Hi, Thank you for your help. I try to include that, but it doesn't work
(arrows are still missing). Where do you you include plt.ion()?  Thanks.

drammock, 2020-07-21:
The problem has been fixed in the current master version of MNE-Python. But even without updating, this should work (putting `plt.ion()` before any plotting calls; typical practice is to put it right after the import statements:

```python
import numpy as np
import matplotlib.pyplot as plt
from mne.datasets import sample
from mne import read_evokeds
from mne.viz import plot_arrowmap
plt.ion()
path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'
condition = 'Left Auditory'
evoked = read_evokeds(fname, condition=condition, baseline=(None, 0))
evoked_mag = evoked.copy().pick_types(meg='mag')
max_time_idx = np.abs(evoked_mag.data).mean(axis=0).argmax()
plot_arrowmap(evoked_mag.data[:, max_time_idx], evoked_mag.info)
```

wsxmxm, 2020-07-21:
Thanks. I have tried the code you provided, but the arrows are still
missing.
I have also updated my MNE to the master version. But the plot_arrowmap(
evoked_mag.data[:, max_time_idx], evoked_mag.info) always causes a memory
error and the restart of the kernel. There are still no arrows.

drammock, 2020-07-21:
To reduce memory usage you could try avoiding the `.copy()` operation:

```python
evoked.pick_types(meg='mag')
max_time_idx = np.abs(evoked.data).mean(axis=0).argmax()
plot_arrowmap(evoked.data[:, max_time_idx], evoked.info)
```

I'll look into the `plot_arrowmap` function and see if there are ways to reduce our memory usage, but it might be a couple days.

wsxmxm, 2020-07-21:
Avoiding the .copy() is not enough. There seems too much memory usage.
Great. I look forward to your magic on the plot_arrowmap function. Many
thanks.

drammock, 2020-08-06:
@wsxmxm I finally had time to look into this, and honestly I can't understand why you keep hitting memory errors.  The code sample I entered above uses less than 200 MiB in my testing, and after a fair bit of profiling I can't see any obvious way to trim that down:

- about 3/4ths of the memory usage is just from loading the libraries
- only about 37 MiB are allocated during the call to `plot_arrowmap()`
- of that 37 MiB, about 29 MiB are allocated by matplotlib for the figure creation and tight_layout.

In sum, there is very little to be done here to reduce memory usage here.  Since your system has 3.7 GB of RAM, this really ought to be possible for you to run (as long as you don't have several other programs running at the same time, using up memory).

wsxmxm, 2020-08-27:
Hi, many thanks for your double checking. I still have the strange memory
usuage problem. I will try to check on another computer. Thank you again
for helping me.

PR #8034 (2020-07-20): VIZ, BUG: delay plt_show until after quiver
closes #8032