mne-tools/mne-python #8036: trying to import edf file using mne on linux.

harmeet9307, 2020-07-21:
Please check the below code and suggest edit. When I run the same in windows, it works. the package versions are identical. 

_**data  = mne.io.read_raw_edf('eeg1.edf',preload=True,exclude='meas_date')

//
Extracting EDF parameters from /wsu/home/ha/ha82/ha8276/Desktop/EEG/eeg1.edf...
EDF file detected

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-11-8c2213f7e7ee> in <module>
----> 1 data  = mne.io.read_raw_edf('eeg1.edf',preload=True,exclude='meas_date')

~/.local/lib/python3.7/site-packages/mne/io/edf/edf.py in read_raw_edf(input_fname, eog, misc, stim_channel, exclude, preload, verbose)
   1179     return RawEDF(input_fname=input_fname, eog=eog, misc=misc,
   1180                   stim_channel=stim_channel, exclude=exclude, preload=preload,
-> 1181                   verbose=verbose)
   1182 
   1183 

<decorator-gen-163> in __init__(self, input_fname, eog, misc, stim_channel, exclude, preload, verbose)

~/.local/lib/python3.7/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, eog, misc, stim_channel, exclude, preload, verbose)
    113         info, edf_info, orig_units = _get_info(input_fname,
    114                                                stim_channel, eog, misc,
--> 115                                                exclude, preload)
    116         logger.info('Creating raw.info structure...')
    117 

~/.local/lib/python3.7/site-packages/mne/io/edf/edf.py in _get_info(fname, stim_channel, eog, misc, exclude, preload)
    375     misc = misc if misc is not None else []
    376 
--> 377     edf_info, orig_units = _read_header(fname, exclude)
    378 
    379     # XXX: `tal_ch_names` to pass to `_check_stim_channel` should be computed

~/.local/lib/python3.7/site-packages/mne/io/edf/edf.py in _read_header(fname, exclude)
    362     logger.info('%s file detected' % ext.upper())
    363     if ext in ('bdf', 'edf'):
--> 364         return _read_edf_header(fname, exclude)
    365     elif ext in ('gdf'):
    366         return _read_gdf_header(fname, exclude), None

~/.local/lib/python3.7/site-packages/mne/io/edf/edf.py in _read_edf_header(fname, exclude)
    543         century = 2000 if year < 50 else 1900
    544         meas_date = datetime(year + century, month, day, hour, minute, sec,
--> 545                              tzinfo=timezone.utc)
    546 
    547         header_nbytes = int(fid.read(8).decode())

ValueError: month must be in 1..12

//**_

cbrnr, 2020-07-21:
Why do you `exclude='meas_date'`? You did not specify this argument in the first line, but it appears in the traceback.

harmeet9307, 2020-07-21:
I was trying something but dint work. Even when i do not mention it, the error says “ month should be in 1...12” 
But the same file and package and commands work just fine in Windows. 

Regards
Harmeet Singh Chhibber

cbrnr, 2020-07-21:
Alright. Could you share the problematic file with me?

harmeet9307, 2020-07-21:
Sure, it wont attach here. Here is a link to the directory with the files. Any eeg.edf can be used. The issue is the same with all. 
link: https://zenodo.org/record/2547147#.XxbzN-cpBhE

cbrnr, 2020-07-21:
I tried [eeg1.edf](https://zenodo.org/record/2547147/files/eeg1.edf?download=1) and it loads without any problems on macOS. I can test on Linux tomorrow - if anyone on Linux wants to beat me to it feel free to let me know.

harmeet9307, 2020-07-21:
Yeah it works fine with windows too. I am not even sure if its linux or just my vm,

cbrnr, 2020-07-21:
Might also be an older version on Linux? Can you post the output of `mne.sys_info()` for both platforms?

harmeet9307, 2020-07-21:
Platform:      Linux-3.10.0-1127.10.1.el7.x86_64-x86_64-with-centos-7.8.2003-Core
Python:        3.7.7 (default, Mar 26 2020, 15:48:22)  [GCC 7.3.0]
Executable:    /wsu/el7/pre-compiled/python/3.7/envs/tensorflow_gpuenv_v2/bin/python
CPU:           x86_64: 40 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.23.1
numba:         Not found
nibabel:       3.1.0
cupy:          Not found
pandas:        1.0.3
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

for the linux and windows is below

Platform:      Windows-10-10.0.19041-SP0
Python:        3.7.3 (default, Mar 27 2019, 17:13:21) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Users\harme\Anaconda3\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:        7.8 GB

mne:           0.20.0
numpy:         1.18.2 {blas=C:\\projects\\numpy-wheels\\numpy\\build\\openblas_info, lapack=C:\\projects\\numpy-wheels\\numpy\\build\\openblas_lapack_info}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.23.1
numba:         0.43.1
nibabel:       Not found
cupy:          Not found
pandas:        0.25.1
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

drammock, 2020-07-21:
I can successfully read & preload the file @cbrnr linked to, on Linux (Xubuntu 20.04), using both MNE-Python 0.20.7 and using current master version of MNE-Python, as `raw = mne.io.read_raw_edf('eeg1.edf',preload=True)`.

```console
$ mne sys_info
Platform:      Linux-5.4.0-40-generic-x86_64-with-glibc2.10
Python:        3.8.3 (default, May 19 2020, 18:47:26)  [GCC 7.3.0]
Executable:    /opt/miniconda3/envs/mne/bin/python
CPU:           x86_64: 16 cores
Memory:        125.7 GB

mne:           0.20.7
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=Qt5Agg}

sklearn:       0.22.1
numba:         0.49.1
nibabel:       3.1.0
cupy:          Not found
pandas:        1.0.3
dipy:          1.1.1
mayavi:        4.7.2.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.24.2
vtk:           9.0.0
```

cbrnr, 2020-07-21:
Thanks @drammock. @harmeet9307 maybe the file hasn't been downloaded correctly on Linux? Can you check the MD5 and compare if you get the same checksum on all three platforms? It should be `db6659835476f60bcf9a6b20f63c641d`.

harmeet9307, 2020-07-21:
Yes, I checked that. It isnt the issue. 

Regards
Harmeet Singh Chhibber

cbrnr, 2020-07-21:
Although this probably isn't the issue, but can you update to the latest MNE release? And if that doesn't work can you try master?

Can you do a `%debug` after you get the error and let us know the values of `month` and the other date-related ones?

basophobic, 2020-07-22:
I have also tried loading the edf files on my opensuse linux distro. All works fine.

```
>>> data  = mne.io.read_raw_edf('/home/koutras/Temp/eeg10.edf',preload=True)

Extracting EDF parameters from /home/koutras/Temp/eeg10.edf...
EDF file detected
Setting channel info structure...
Creating raw.info structure...
Reading 0 ... 1389311  =      0.000 ...  5426.996 secs...
```
and my system configuration:
```
$ mne.sys_info()

Platform:      Linux-5.7.7-1-default-x86_64-with-glibc2.10
Python:        3.7.5 (default, Oct 25 2019, 15:51:11)  [GCC 7.3.0]
Executable:    /home/koutras/anaconda3/envs/tf2_env/bin/python
CPU:           x86_64: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.7
numpy:         1.17.3 {blas=openblas, lapack=openblas}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=module://ipykernel.pylab.backend_inline}
sklearn:       0.21.3
numba:         0.50.1
nibabel:       3.1.1
cupy:          Not found
pandas:        0.25.3
dipy:          1.1.1
mayavi:        Not found
pyvista:       0.25.3
vtk:           9.0.1
```

cbrnr, 2020-07-22:
Thanks @basophobic - it looks like this could be a misconfigured MNE setup. @harmeet9307 can you try to install MNE from scratch as described [here](https://mne.tools/stable/install/mne_python.html)?

harmeet9307, 2020-07-22:
Sure, I tried to reinstall mne. Still get the same error. 

Regards
Harmeet Singh Chhibber

drammock, 2020-07-22:
> Can you do a `%debug` after you get the error and let us know the values of `month` and the other date-related ones?

This is still the logical next step in debugging, and since none of us can replicate your error, you'll need to do this yourself @harmeet9307

Since it looks like you're not using ipython, the equivalent of `%debug` would be `import pdb; pdb.pm()` to enter the debugger immediately after the error occurs. from there you should be able to view the values of the variables `year`, `month`, etc.

harmeet9307, 2020-07-27:
I figured it out. there were some issues with the edf files I downloaded. Works fine with another batch of files. Thank you are all the help

closed 2020-07-27 by harmeet9307