mne-tools/mne-python #8039: Getting Error: "channel position not present in the DigMontage" when assigning a montage in MNE

balandongiv, 2020-07-22:
#### Describe the bug
The objective is to assign '`GSN-HydroCel-128`' montage to the raw data. The raw data was collected using `HydroCel GSN 128 1.0`.

However, the compiler return an error:

**`ValueError: DigMontage is only a subset of info. There are 1 channel position not present in the DigMontage. The required channels are: ['E129']`**

However, the mentioned channel `'E129'` does exist as shown from the  `print(raw.ch_names)` below

`['E1', 'E2', 'E3', 'E4', 'E5', 'E6', 'E7', 'E8', 'E9', 'E10', 'E11', 'E12', 'E13', 'E14', 'E15', 'E16', 'E17', 'E18', 'E19', 'E20', 'E21', 'E22', 'E23', 'E24', 'E25', 'E26', 'E27', 'E28', 'E29', 'E30', 'E31', 'E32', 'E33', 'E34', 'E35', 'E36', 'E37', 'E38', 'E39', 'E40', 'E41', 'E42', 'E43', 'E44', 'E45', 'E46', 'E47', 'E48', 'E49', 'E50', 'E51', 'E52', 'E53', 'E54', 'E55', 'E56', 'E57', 'E58', 'E59', 'E60', 'E61', 'E62', 'E63', 'E64', 'E65', 'E66', 'E67', 'E68', 'E69', 'E70', 'E71', 'E72', 'E73', 'E74', 'E75', 'E76', 'E77', 'E78', 'E79', 'E80', 'E81', 'E82', 'E83', 'E84', 'E85', 'E86', 'E87', 'E88', 'E89', 'E90', 'E91', 'E92', 'E93', 'E94', 'E95', 'E96', 'E97', 'E98', 'E99', 'E100', 'E101', 'E102', 'E103', 'E104', 'E105', 'E106', 'E107', 'E108', 'E109', 'E110', 'E111', 'E112', 'E113', 'E114', 'E115', 'E116', 'E117', 'E118', 'E119', 'E120', 'E121', 'E122', 'E123', 'E124', 'E125', 'E126', 'E127', 'E128', 'E129', 'Resp.Temp', 'Resp.Pressure', 'ECG', 'Body.Position', 'Resp.Effort.Che', 'Resp.Effort.Abd', 'EMG']`

**Full error report:**

```
  File "<decorator-gen-20>", line 21, in set_montage
  File "\lib\site-packages\mne\io\meas_info.py", line 180, in set_montage
    _set_montage(info, montage, match_case, on_missing)
  File "\lib\site-packages\mne\channels\montage.py", line 734, in _set_montage
    raise ValueError(missing_coord_msg)

ValueError: DigMontage is only a subset of info. There are 1 channel position not present in the DigMontage. The required channels are: ['E129']
```

#### Steps to reproduce

The steps can be reproduced via the snippet below

```
    filename = r'S17_5.mff' # The mff file can be downloaded via the link provided
    raw = mne.io.read_raw_egi(filename, preload=True)
    print(raw.ch_names)
    montage = mne.channels.make_standard_montage('GSN-HydroCel-128')
    raw.set_montage(montage, match_case=False)
```

#### Additional information

System:
Windows 10
anaconda3
mne-0.21.dev0 
Python 3.7
[Link to download S17_5.mff ](https://drive.google.com/drive/folders/15lAjuEPC2iSbBKBSy8y11Q0Fs24cnSd1?usp=sharing)

balandongiv, 2020-07-22:
Not sure whether this related to [#8025 ],[#7615 ]

agramfort, 2020-07-22:
try

import mne
filename = r'S17_1.mff' # The mff file can be downloaded via the link
provided
raw = mne.io.read_raw_egi(filename, preload=True)
print(raw.ch_names)
montage = mne.channels.make_standard_montage('GSN-HydroCel-129')
montage.ch_names[-1] = 'E129'
raw.set_montage(montage, match_case=False)

or

import mne
filename = r'S17_1.mff' # The mff file can be downloaded via the link
provided
raw = mne.io.read_raw_egi(filename, preload=True)
print(raw.ch_names)
montage = mne.channels.make_standard_montage('GSN-HydroCel-256')
raw.set_montage(montage, match_case=False)

>

balandongiv, 2020-07-22:
Thanks for the feedback @agramfort . Both approach return no error. But, it lead to next question on why use `'GSN-HydroCel-129' `and `'GSN-HydroCel-256'` despite the fact that the data was collected from `HydroCel GSN 128 `? But, since you are the expert, I believe the suggestion should be correct then.

closed 2020-07-22 by balandongiv

drammock, 2020-07-22:
@balandongiv if you look more closely at the original error, it says that `E129` is not present **in the DigMontage** (what you demonstrated with `raw.ch_names` is that the channel is in fact **in the data**).  Because your data has 129 data channels, the 128-channel montage will not know where to locate the last channel.  In the 129-channel montage the final channel is labelled `Cz`, which is why @agramfort suggested these lines:

```python
montage = mne.channels.make_standard_montage('GSN-HydroCel-129')
montage.ch_names[-1] = 'E129'
```

But using the 256-channel montage also works, because it is OK if there are channels in the *montage* that are not present in the *data*.  What causes errors like this is when there are channels in the *data* that are not in the *montage*.  Incidentally, another approach that probably would have worked would be to use the 129-channel montage, but to rename the *data* channel from `E129` to `Cz` (instead of renaming the last *montage* channel from `Cz` to `E129`)

balandongiv, 2020-07-23:
> @balandongiv if you look more closely at the original error, it says that `E129` is not present **in the DigMontage** (what you demonstrated with `raw.ch_names` is that the channel is in fact **in [...]
Thanks for the detail explanation @drammock. Now it really make sense to me.  

> ```python
> montage = mne.channels.make_standard_montage('GSN-HydroCel-129')
> montage.ch_names[-1] = 'E129'
> ```
>
> But using the 256-channel montage also works, because it is OK if there are channels in [...]