mne-tools/mne-python #8040: conda environment.yml broken?

cbrnr, 2020-07-22:
Currently, all CIs using `conda` with our `environment.yml` are broken because of unsatisfiable dependencies. It has something to do with `traits`, which is available as 5.2.0, but some other package (`pyface`?) wants >=6.0.0. I tested this locally, but everything works fine, so maybe the problematic packages have already been updated upstream. I can't restart Travis jobs for some reason, maybe @sappelhoff can you do it in #8037?

Related issue: I noticed that `pyqt` 5.9.2 gets installed automatically as a dependency by `conda`, but then we overwrite it with a newer version of `PyQt5` available in `pip`. This could be problematic and might explain issues related to PyQt that pop up from time to time.

sappelhoff, 2020-07-22:
[image]

https://travis-ci.org/github/mne-tools/mne-python/builds/710508815

cbrnr, 2020-07-22:
Now the first jobs have a different error:

```
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: unrecognized arguments: -n
```

cbrnr, 2020-07-22:
There doesn't seem to be a `-n` argument (`pytest -m "${CONDITION}" --tb=short --cov=mne -n 1 ${USE_DIRS}`).

GuillaumeFavelier, 2020-07-22:
`-n` is used by the `pytest-xdist` plugin. It's not installed?

cbrnr, 2020-07-22:
Looks like not. Where is the requirements file that contains this package?

cbrnr, 2020-07-22:
@sappelhoff could you maybe also restart the failing Azure job?

sappelhoff, 2020-07-22:
done, but I can only re-run the failed ones. Can only the PR owner or an admin re-run? I think you and I should have same permissions @cbrnr

cbrnr, 2020-07-22:
The failed ones are OK for now. I didn't find the restart button in Travis (although I was logged in and I know where the button usually is - now there is just a "More options" button). I haven't figured out how to restart Azure jobs, I'm logged in and now I see only a "Cancel" button for the running job.

sappelhoff, 2020-07-22:
>  I haven't figured out how to restart Azure jobs, I'm logged in and now I see only a "Cancel" button for the running job.

you do it from the GitHub UI, go to "details" and then there is an expanded dropdown menu on the left

cbrnr, 2020-07-22:
This page? I don't have a dropdown menu.

[image]

sappelhoff, 2020-07-22:
This is the one I mean:

[image]

and it shows handles when you can restart something. you can't do anything for running or succeeded jobs with simple maintainer permissions.

cbrnr, 2020-07-22:
Fixed upstream.

closed 2020-07-22 by cbrnr