mne-tools/mne-python #8044: Undesireable output of the to_data_frame

balandongiv, 2020-07-22:
#### Describe the bug
The idea is to produce the output with respect to set `time_format` parameter  under the `to_data_frame `method. However, setting the time_format to either  `None, 'timedelta', 'datetime' and 'ms'` produce an incremental plus 1 integer instead of the scaled time (e.g., 0, 0.001,0.002) for a signal that sampled in 1000Hz.

#### Steps to reproduce

* filename='S17_5.mff' # The mff file is accessible via download link provided below
```
raw = mne.io.read_raw_egi(filename, preload=True)
print(raw.to_data_frame(picks=['E1']))
```

Return the following

```
Reading 0 ... 300936  =      0.000 ...   300.936 secs...
 	time            E1
0            0  15372.021811
1            1  15375.733752
2            2  15371.825459
3            3  15367.826391
4            4  15363.235307

```

```
raw = mne.io.read_raw_egi(filename, preload=True)
print(raw.to_data_frame(picks=['E1'],time_format='ms'))
```

Return the following

```
          time            E1
0            0  15372.021811
1            1  15375.733752
2            2  15371.825459
3            3  15367.826391
4            4  15363.235307
```

```

raw = mne.io.read_raw_egi(filename, preload=True)
print(raw.to_data_frame(picks=['E1'],time_format='timedelta'))
```

Return the following
```
                  time            E1
0             00:00:00  15372.021811
1      00:00:00.001000  15375.733752
2      00:00:00.002000  15371.825459
3      00:00:00.003000  15367.826391
4      00:00:00.004000  15363.235307
```

#### Expected results

At least, the column time should look

`[0.    0.001 0.002 0.003 0.004 0.005 0.006 0.007 0.008 ]`

#### Actual results

#### Additional information
System:

Windows 10
anaconda3
mne-0.21.dev0 
Python 3.7
[Link to download S17_5.mff ](https://drive.google.com/drive/folders/15lAjuEPC2iSbBKBSy8y11Q0Fs24cnSd1?usp=sharing)

hoechenberger, 2020-07-22:
So what you you're looking for is, essentially, support for `time_format='s'`?

drammock, 2020-07-22:
@hoechenberger is correct that this is not a bug. The problem here is that `time_format=None` is not the default, so you have to explicitly pass `None` in order to get the equivalent of `time_format='s'`.  Eventually we plan to change this to default to `None`, but it will need to go through a deprecation cycle.

```python
import os
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=5)
raw.to_data_frame(picks=['MEG 0122'], time_format=None).head()                          
#        time    MEG 0122
# 0  0.000000  144.653322
# 1  0.001665  106.079103
# 2  0.003330   48.217774
# 3  0.004995   67.504884
# 4  0.006660  115.722658
```

closed 2020-07-22 by drammock

balandongiv, 2020-07-23:
Thanks for the clarification