mne-tools/mne-python #8045: Inconsistent topoplot layout across different TFR plotting functions

hoechenberger, 2020-07-22:
#### Describe the bug
@virvw reported some weird-looking topomap figures when plotting TFR data. While I couldn't **exactly** replicate the issues, I **do** find inconsistencies in the figures produced by `plot_joint()`, `plot_topo`, and `plot_topomap`.

#### Steps to reproduce

Load data & calculate power 
```python
import os.path as op
import numpy as np
import matplotlib

import mne
from mne.time_frequency import tfr_morlet
from mne.datasets import somato

matplotlib.use('Qt5Agg')

data_path = somato.data_path()
subject = '01'
task = 'somato'
raw_fname = op.join(data_path, 'sub-{}'.format(subject), 'meg',
                    'sub-{}_task-{}_meg.fif'.format(subject, task))

raw = mne.io.read_raw_fif(raw_fname)
events = mne.find_events(raw, stim_channel='STI 014')
picks = mne.pick_types(raw.info, meg='grad', eeg=False, eog=True, stim=False)

# Construct Epochs
event_id, tmin, tmax = 1, -1., 3.
baseline = (None, 0)
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=baseline, reject=dict(grad=4000e-13, eog=350e-6),
                    preload=True)
epochs.resample(200., npad='auto')

freqs = np.logspace(*np.log10([6, 35]), num=8)
n_cycles = freqs / 2
power, _ = tfr_morlet(epochs, freqs=freqs, n_cycles=n_cycles, use_fft=True,
                      return_itc=True, decim=3, n_jobs=1)
```

Now, let's plot the data. 

### First, via `plot_joint`
```python
power.plot_joint()
```
[image]

When selecting a ROI in that plot, a new window pops up:
[image]

➜ Looks all good so far, if one ignores that the field is chopped off at the back of the head a little.

### Now, let's use `plot_topo`

```python
power.plot_topo()
```
Then, select one of the sensors, and a ROI in the popup. This will produce a figure like this:
[image]

➜ Not only is the topoplot much smaller than in the previous figure, but the colorbar is narrower as well.

### Lastly, let's use power.plot_topomap()
[image]

➜ The entire figure is much smaller than the ones produced with the other methods, and both the topoplot as well as the colorbar are so small that one can barely read this plot at all.

agramfort, 2020-09-02:
@hoechenberger @larsoner @drammock would you have time to look into this one? I would like to see it fixed for next release

hoechenberger, 2020-09-02:
I could have a look this Friday / weekend

hoechenberger, 2020-09-02:
> ➜ The entire figure is much smaller than the ones produced with the other methods, and both the topoplot as well as the colorbar are so small that one can barely read this plot at all.

… and we don't see the helmet shape in the smaller figures.

larsoner, 2020-09-09:
Keep in mind that `plot_topo` should produce different looking plots than `plot_topomap` -- the former uses `layout`s that are different from the physically mapped `topomap` plots, at least for Neuromag MEG data (for things like EEG they might be the same or quite similar).

The head being tiny, however, is a bug that we should fix. Any progress on this @hoechenberger ?

PR #8255 (2020-09-14): MRG, BUG: Fix bugs with TFR topo(map) plotting
Closes #8045

@hoechenberger feel free to test with this slightly abbreviated minimal example:

<details>

```
import os.path as op
import numpy as np

import mne
from mne.time_frequency import tfr_morlet
from mne.datasets import somato

data_path = somato.data_path()
subject = '01'
task = 'somato'
raw_fname = op.join(data_path, 'sub-01', 'meg', 'sub-01_task-somato_meg.fif')
raw = mne.io.read_raw_fif(raw_fname)
events = mne.find_events(raw, stim_channel='STI 014')
picks = mne.pick_types(raw.info, meg='grad', eeg=False, eog=True, stim=False)
event_ [... cut]
closed 2020-09-14 by hoechenberger