mne-tools/mne-python #8053: Incorrect number of EOG events detected when using find_eog_events for data recorded with EGI HydroCel GSN 128 1.0 system

balandongiv, 2020-07-24:
#### Describe the bug
This report is about inconsistent behavior when applying the `mne.preprocessing.find_eog_events()`.

Manually checking the video recorded during the bio signal recording indicated that  there should be more than 1 `EOG` events.
However, the report generated upon applying the `mne.preprocessing.find_eog_events()` gives only 1 `EOG` event detected.

> Now detecting blinks and generating corresponding events
> Found 1 significant peak
> Number of EOG events detected : 1

#### Steps to reproduce

The above report can be reproduced via the code snippet  below:

```
import mne
import matplotlib  
matplotlib.use('Qt5Agg') # Required for PyCharm user

filename = r'S17_3.mff' # The mff file is accessible via download link provided below

raw = mne.io.read_raw_egi(filename, preload=True)
montage = mne.channels.make_standard_montage('GSN-HydroCel-129')
montage.ch_names[-1] = 'E129'
raw.set_montage(montage, match_case=False)
raw.filter(1., 50.)

# Look for EOG event
eog_events = mne.preprocessing.find_eog_events(raw, ch_name='E126,E127')
onsets = eog_events[:, 0] / raw.info['sfreq'] - 0.25
durations = [0.5] * len(eog_events)
descriptions = ['bad blink'] * len(eog_events)
blink_annot = mne.Annotations(onsets, durations, descriptions,
                                  orig_time=raw.info['meas_date'])
eeg_picks = mne.pick_types(raw.info,eeg=True)
# raw.plot(events=eog_events, order=eeg_picks)
raw.plot(events=eog_events)
raw.set_annotations(blink_annot)
```

#### Expected results
There should be more than ONE `EOG` events

#### Additional information
1) System:

Windows 10
anaconda3
mne-0.21.dev0
Python 3.7
[Link to download S17_3.mff ](https://drive.google.com/drive/folders/14R_2A-ceqOFvRkDZH6Skm4CqF5B-K7Co?usp=sharing)

2) Channel selection as replacement of EOG channels

Since the  `128-channel HCGSN v.1.0` does not have dedicated EOG channels (correct me if Im wrong about this), the channel E126 and E127 were selected as surrogate as per usage example in [#4027](https://github.com/mne-tools/mne-python/issues/4027#issue-211059521). However, I am open to any suggestion if there is any other channels combination that can improved the EOG event inspection.

agramfort, 2020-07-24:
did you inspect the data? maybe you have some artifact in the data? this
function is not guaranteed
to work and makes some assumptions that may not be fulfilled in your data.

>

balandongiv, 2020-07-24:
Thanks for the prompt reply @agramfort . 
May I know what kind of artifact you have in mind. FYI, I have tested other `mff` files from other subjects as well, and all of the testing return little EOG event. Hence, my guess maybe to aforementioned artifact maybe not the source of problem. But to confirm, is it sufficient/advisable to use the E126 and E127 here?

Since this function is not guaranteed, then is it advisable to rely on `mne.preprocessing.ICA `?

Appreciate for any advice.

agramfort, 2020-07-24:
I don't know your data but to see of  E126 and E127 can work they should
contain clear blink patterns

>

balandongiv, 2020-07-24:
Thanks for the feedback @agramfort . But, as of this stage, I believe there is something missing with my setup as the number of eog event detected is quite small. Anyhow, I will update the thread once I got thing working.

balandongiv, 2020-07-24:
Other approach to surrogate absence of EOG channel is by creating a new virtual channel. This is possible using  `mne.set_bipolar_reference` with its usage detail accessible via the [link](https://mne.tools/stable/generated/mne.set_bipolar_reference.html).

Say, we would like to consider only the left side of the eye only to detect eye blinks, then we can use the channel `E25` and `E127` to create a new virtual channel with the name `EVirt`. 

`raw=mne.set_bipolar_reference(raw, ['E25'],['E127'],['EVirt'])`

or the combination of `E8` and `E126`

Then, set the new virtual channel as the type of `EOG`

`raw=set_channel_types({'EVirt': 'eog'})`

Then, the EOG event can be search simply by:

`eog_events =eog.find_eog_events(raw, reject_by_annotation=True, thresh=0.0001, verbose=None)`

This approach should work for data collected with EGI HydroCel GSN 128 1.0 system, atleast for my data. However, I yet to compare manually the eog_events with those with eye blink detected with video/image captured.

balandongiv, 2020-07-24:
Just to add,
using different channel combination to create the virtual channel result in diffrent number of eog event detected

1. `E8 & E126` resulting in 98 EOG events
2. `E14 & E126` resulting in 122 EOG events
3. `E21 & E127` resulting in 110 EOG events
4. `E25 & E127` resulting in 98 EOG events

Is this mean we need to manually inspect the event as well

agramfort, 2020-07-24:
try this:

```
import mne
# import matplotlib
# matplotlib.use('Qt5Agg') # Required for PyCharm user

filename = r'S17_3.mff'  # The mff file is accessible via download link provided below

raw = mne.io.read_raw_egi(filename, preload=True)
montage = mne.channels.make_standard_montage('GSN-HydroCel-129')
montage.ch_names[-1] = 'E129'
raw.set_montage(montage, match_case=False)
raw.info['bads'] = ['E129', 'E114', 'E88']  # mark bad channels
raw.filter(1., 40.)
raw.crop(0, 595)  # crop huge artifact at the end

raw.plot_psd()

# Look for EOG event
eog_events = mne.preprocessing.find_eog_events(raw, ch_name='E124')
onsets = eog_events[:, 0] / raw.info['sfreq'] - 0.25
durations = [0.5] * len(eog_events)
descriptions = ['bad blink'] * len(eog_events)
blink_annot = mne.Annotations(onsets, durations, descriptions,
                              orig_time=raw.info['meas_date'])
# eeg_picks = mne.pick_types(raw.info, eeg=True)
# raw.plot(events=eog_events, order=eeg_picks)
raw.plot(events=eog_events)
raw.set_annotations(blink_annot)
```

you did have a big artifact at the end of the file.

I also marked a few obvious bad channels

agramfort, 2020-07-24:
I also filtered below 40 has you have a huge line noise artifact at 50Hz that remained with a low pass at 50

closed 2020-07-24 by agramfort

balandongiv, 2020-07-25:
Hi @agramfort , thanks for the sample code, I really appreciate it.

balandongiv, 2020-07-27:
Hi @agramfort, I notice that you used  `ch_name='E124'` to localise the EOG event. I cross check with the 128-sensor HydroCel GSN recording montage, the E124 correspond to  `F4`. While other have proposed the used of [Fp1/Fp2](https://www.hindawi.com/journals/tswj/2014/259121/) or VEOG and HEOG which I believe any of the combination of EEG channel on the face that correspond to [EVirt](https://github.com/mne-tools/mne-python/issues/8053#issuecomment-663530725), but you choose to use F4 instead. May I know if there is good motivation behind this that I am not aware of? Thanks for the time entertain this question.

agramfort, 2020-07-27:
I looked at the data at it contains clear blink artfifacts

balandongiv, 2020-07-28:
For future reader as well reference for me later.

As mentioned by @agramfort, he marked few bad channels namely 'E114', 'E88'. The artifact can be observed as depicted in the picture which is accessible
[image]
  [via the link](https://imgur.com/gallery/hcJzHBY).

balandongiv, 2020-07-28:
[image]