mne-tools/mne-python #8060: ENH: Add support for writing other formats

larsoner, 2020-07-26:
@rob-luke wants to add support for writing SNIRF files. We could support this internally easily via file-extension `*.snirf` triage. WDYT @agramfort ?

There are also some potentially low-hanging fruit in terms of supporting writing to other formats using external libraries, at least BrainVisian (pybv) EDF (pyEDF, maybe edflib?) and CTF (pyCTF). I don't see too much work in supporting these since in theory we "just" hook up our structures to the right places to their writers, then rely on them to write the correct data. I don't see this as too much additional maintenance burden for us since most of the heavy lifting will be done by those other libraries.

Thoughts?

agramfort, 2020-07-27:
I don't know these external packages: if they are released regularly on
pypi, if they are unit tested / maintained, if license is ok etc.
I just fear that users will complain to us if these packages break.

can we first try with a write_raw_fnirs in mne-nirs and maybe reconsider in
a few months after seeing how smoothly this goes?

it's just once it's in MNE it tends to stay for a long time and we'll have
to maintain it.

>

rob-luke, 2020-07-28:
> can we first try with a write_raw_fnirs in mne-nirs and maybe reconsider in a few months after seeing how smoothly this goes?

Sure I will put this in mne-nirs

sappelhoff, 2020-08-25:
Just fyi: I maintain **[pybv](https://github.com/bids-standard/pybv/)** with help from @choldgraf, and @cbrnr (at least he made a release and a few PRs :slightly_smiling_face: ) ... @hoechenberger has made the corresponding conda-forge package work, and apart from that there is one more non-MNE contributor :-)

We have a BSD-3-Clause License, a pypi and conda forge package, and unit tests. However, we do not regularly release, because the writer currently already supports all features that the contributor and maintainer team need.

Note: some parts of the code in pybv stem from @palday's "philistine" package. As far as I can see it, @palday also maintains [pyedf](https://github.com/bids-standard/pyedf/)

Yet with regards to EDF writing, the development looks more active in [pyedflib](https://github.com/holgern/pyedflib) ... with the drawback that that writer is not pure python, it depends on the [edflib C library](https://gitlab.com/Teuniz/EDFlib)

cbrnr, 2020-08-25:
MNELAB supports exporting to BrainVision using pybv, EDF/BDF using pyEDFlib, and EEGLAB SET with custom code. I'd be happy to move this code to MNE so that everyone can benefit from this functionality (but I don't know when I will find the time to do that, currently I have too many other things on my plate).

palday, 2020-08-25:
I have become the defacto maintainer of pyedf, but I inherited a codebase completely without tests, so until I get around to writing those, I'm very picky about PRs (as the currently open ones will know). 

If MNELAB already wraps pybv, using that code is the way to go, but otherwise philistine's functionality was more strongly intertwined with MNE than pybv, so some time travel to pybv's roots might be useful. 

My recommendation: add a separate `export` or use explicit `write_raw_fileformat` patterns to make it clear that some things in MNE's internal FIFF representation can be lost when exporting. Moreover, a roundtrip read-write cycle for any given format may be lossy (e.g. BrainVision's support for channel specific filters will be lost).

larsoner, 2021-04-02:
From Discord discussion between me, @britta-wstnr, @hoechenberger, @drammock, and @agramfort during office hours we hatched the following plan:

- Add `mne.io.base.BaseRaw.export`, perhaps with the signature `def export(fname, fmt='auto')` where `fmt` determines which format is being written. `'auto'` means "determine from the extension".
- Internally, the job of each `fmt` flavor of `export` is to wrap our `info` / annotations / data as well as possible, and use some external dependency (`pybv`, `pyedf`, etc.) to do the writing
- MNE-Python will not guarantee losslessness in `.export` -- we only do this in `.save` for FIF -- and document clearly with a warning at the top of the docstring that the export is a "best effort" but not guaranteed to preserve all information
- Our tests are basic round-trip (basically test some or all of the existing files, hopefully just < 10 sec of added testing time per format), leaving completeness testing to the external libraries

A possible PR plan:

1. In the first PR, I propose we add support for BV and EDF via a copy-paste code dump from MNELAB. @cbrnr would you be interested in leading this since it's presumably mostly your code?
2. For EEGLAB format, I we should probably to move the custom MNELAB EEGLAB-exporting code over to the new `eeglabio` module as much as possible, then make a second PR to MNE-Python to export raw data in MNE-Python.
3. Concurrent or after the second PR, @jackz314 #9192 can be modified to make a `BaseEpochs.export` with an API that matches that of `BaseRaw.export` as much as possible. Otherwise I think #9192 already fits pretty well with the plan outlined for `BaseRaw` above.

larsoner, 2021-04-02:
@rob-luke the above would also give you an avenue to support SNIRF writing

rob-luke, 2021-04-02:
Wonderfull. Thanks. I've put this on hold until the society SNIRF validator is up and running, then I'll get back to it.

cbrnr, 2021-04-03:
I haven't had time to comment so I'm late to the party. I previously said that a separate IO package would be nice. However, I was thinking of a package which supports multiple file formats, not just one. I'm now not sure if this is a good idea, or if having various dedicated packages is the better solution.

Regarding EEGLAB export, the code is simple enough that it could be directly integrated into MNE. I know this seems to contradict my previous comments, but there I was assuming we were talking about a multi-format package. Supporting some formats out of the box without requiring extra dependencies would be a good thing. I can take care of the PR because I can re-use code I've already written for MNELAB.

agramfort, 2021-04-03:
> I can take care of the PR because I can re-use code I've already written for MNELAB.

Go for it

jackz314, 2021-04-03:
> I haven't had time to comment so I'm late to the party. I previously said that a separate IO package would be nice. However, I was thinking of a package which supports multiple file formats, not [...]

Does this mean the new `eeglabio` package is now obsolete? Also, I want to make sure that the other issues with integrating export directly into MNE are still considered here (maintenance, testing, etc.).

agramfort, 2021-04-03:
please coordinate with @cbrnr

cbrnr, 2021-04-06:
@jackz314 I think adding EEGLAB export directly to MNE is feasible in terms of maintenance and testing. Of course, the same could be said about BrainVision export, but this format is officially supported by BIDS so it makes sense to have a dedicated package with all bells and whistles. The EEGLAB format is neither supported by BIDS nor is it anything new - it's just a .mat file (we won't support the two-file format with a .set/.fdt couple I assume). I think we only have to test I/O roundtrip within MNE and not EEGLAB, meaning that our own reader will need to read what we export. Therefore, we don't need any MATLAB installations on our CIs.

Of course if you think having a separate package is still the better solution, I am not going to talk you out of it. I just think it's not necessary.

Finally, I'm not sure who is involved in setting up the new export infrastructure. We can either wait for this to be implemented, or we can implement it together with EEGLAB support. @larsoner what is the plan regarding this API enhancement?

larsoner, 2021-04-06:
> Therefore, we don't need any MATLAB installations on our CIs... Of course if you think having a separate package is still the better solution, I am not going to talk you out of it. I just think it's not necessary.

I implemented Octave-based CI recently to make sure long-name support worked and it wasn't so bad:

https://github.com/mne-tools/mne-matlab/blob/master/.github/workflows/moxunit.yml
https://github.com/mne-tools/mne-matlab/blob/master/matlab/test/test_ex_read_raw.m#L30-L51

Ideally the EEGLAB export could start building up something similar by doing an EEGLAB install somehow and making sure the files can be read.

+1 for a separate package, even if the code is small. I think it's more maintainable and a cleaner split in the long term.

> Finally, I'm not sure who is involved in setting up the new export infrastructure. We can either wait for this to be implemented, or we can implement it together with EEGLAB support. @larsoner what is the plan regarding this API enhancement?

Look a few posts up, I suggested a PR plan up there:

https://github.com/mne-tools/mne-python/issues/8060#issuecomment-812645784

cbrnr, 2021-04-07:
Hm, if we implement a public `export` API but don't provide a single format, I don't think that this makes a lot of sense. If we really go the all external route, people could just use the dedicated export functions in those packages (they only need to pass `raw` anyway). I'd still prefer `raw.export` and some simple export filters implemented in MNE.

agramfort, 2021-04-07:
fine with me

>

hoechenberger, 2021-04-07:
I like the idea of external packages that we can tap into with a single, consistent  user-facing public API (i.e., `export()`), though...

cbrnr, 2021-04-07:
Yes, I agree that `raw.export` is nice, but not if we don't provide any implementation. I'm arguing that we could ship simple export filters such as EEGLAB and leave the more complicated ones for external packages.

agramfort, 2021-04-07:
ok. If the eeglab export is support simple I could live with that

>

jackz314, 2021-04-07:
> Yes, I agree that `raw.export` is nice, but not if we don't provide any implementation. I'm arguing that we could ship simple export filters such as EEGLAB and leave the more complicated ones for external packages.

I think this depends on what the maintainers define as simple and what are the future plans. As of right now, the code is pretty simple, the code I used to export is only about 200 - 300 lines in total, but maybe in the future more properties/functionalities are added to the export logic, then it could become more bloated. For example, the original export code from @cbrnr used for eeglab export was only about 20 lines, but since our lab wanted access to more fields in the exports such as channel locations, the code had to grow by quite a lot.

I would say that maybe for now it's fine to just embed the export logic into MNE, since maintaining separate packages for small codebases might be overkill and could introduce other synchronization problems between packages (although unlikely). At the same time, embedding the logic now might mean more code in the future that could introduce maintenance problems, and it would also "stand out" from all the other formats that have external packages.

When I made #9204 I didn't think about external packages at all, since I considered the functionalities to be relatively simple. But now after the discussions here and on #9204, I'm not sure which is better. At the end of the day, I'll respect whatever consensus that's reached, and the code/package is ready for both cases.

sappelhoff, 2021-04-07:
I agree with much of what has been said. My personal favourite option would be to have:

- separate repositories for "writing data", and these repositories should be independent from MNE
- For example, `pybv` provides an API that does not require you to pass an MNE `raw` object --> the package is usable also by non-MNE users
- We can have a similar package `pyeeglab` (or whatever) and can host it under mne-tools (or a different org)
- Similarily for EDF, etc.

Then in MNE, I really like the `export` functionality (with well documented caveats that - unlike for FIF - this is not a 100% roundtrip for metadata)

- That export functionality would interface from `raw` objects to the API of the respective packages (`pybv`, `pyeeglab`, `pyedf`, ...) 
- when a user calls `raw.export(fpath="my-eeg-file.vhdr", format="auto")`, MNE would check that `pybv` is installed and tell the user otherwise

cbrnr, 2021-04-07:
Good points, I think it is really important that external packages do not use MNE objects but work with simpler more generic data structures such as NumPy arrays. Then `raw.export` really has added value - I didn't think of that.

Therefore, it's probably best to go ahead with a separate EEGLAB export package.

agramfort, 2021-04-07:
>
> Therefore, it's probably best to go ahead with a separate EEGLAB export
> package.
>
+1

>

jackz314, 2021-04-08:
I updated #9192 and [eeglabio](https://github.com/jackz314/eeglabio) to remove MNE dependencies.

cbrnr, 2021-04-08:
The export function in your package consists of 22 lines - you mentioned that it is now 200-300 lines. Is this the most recent code? Also, you are still passing an MNE object. Furthermore, I think we should not create an EEG variable anymore since the latest EEGLAB file format doesn't use it anymore.

jackz314, 2021-04-08:
> The export function in your package consists of 22 lines - you mentioned that it is now 200-300 lines. Is this the most recent code? 

Export for raw by itself should be about 20 lines, but the util functions (for chanlocs) are longer, and when I mentioned 200-300 lines, I meant the total lines (raw + epochs + utils, etc).

> Also, you are still passing an MNE object.

The last two commits shouldn't pass MNE objects anymore, maybe it wasn't updated on the PR due to the changes requested?

>  Furthermore, I think we should not create an `EEG` variable anymore since the latest EEGLAB file format doesn't use it anymore.

I'm not sure if this breaks on older EEGLAB versions, if not, I can get rid of it.

hoechenberger, 2021-04-08:
> I'm not sure if this breaks on older EEGLAB versions, if not, I can get rid of it.

I remember reading that it's backward-compatible.

jackz314, 2021-04-08:
I implemented a multi-format export structure in the [last commit](https://github.com/mne-tools/mne-python/pull/9192/commits/2d8f1c809fba0a17b613ab8ce149d1b19a483234). I assumed the format will just be the file's extension (e.g., `set`). I'm not sure if using the actual names would be more appropriate here (e.g., `eeglab` instead of `set`).

jackz314, 2021-04-08:
> I remember reading that it's backward-compatible.

I just updated eeglabio to remove the EEG variable. But do you know where this backward compatibility is mentioned? I just want to make sure that it's safe.

cbrnr, 2021-04-08:
Right here: https://github.com/mne-tools/mne-python/issues/8864#issuecomment-779375469

larsoner, 2021-04-19:
We have a plan and it's in progress so I'll close this

closed 2021-04-19 by larsoner