mne-tools/mne-python #8065: MNE plot window in dark mode

balandongiv, 2020-07-28:
Lately, there has been a lot of hype surrounding dark mode, a simple toggle that lets user to change the background color of an app window to black. There are several benefit of dark mode including reduce eye strain and even look amazing. But, the former is crucial as most researcher (at least me) spent a considerable amount of time visually analysing the biosignal graph using plot(). This especially true during the pre processing stage.

#### Describe your solution
Maybe there can be an option to toggle the window plot to change the background color to black, while the text and graph line into contrasting color such as white.

drammock, 2020-07-28:
This is already planned and will be part of #7955

balandongiv, 2020-07-28:
That great. Eager to try this new feature.

closed 2020-07-28 by balandongiv