mne-tools/mne-python #8067: ICA.find_bads_ecg() accepts "method", find_bads_eog() accepts "measure"

hoechenberger, 2020-07-29:
`ICA.find_bads_ecg()` has a kwarg `method : {'ctps', 'correlation'}`; in contrast, `find_bads_eog()` uses `measure : 'zscore' | 'cor'`

- The kwarg for selectiong of the method to use is once called `method`, and once `measure`
- Correlation is once selected by passing `'correlation'`, and once via `'cor'`
- The specification of valid values is once done via `key : {val1, val2}`, and once via `key : val1 | val2`

How can we make things more consistent?

agramfort, 2020-07-29:
I would make it so that find_bads_eog matches the API of find_bads_ecg. So rename measure to method and require correlation and not cor. But I would write it as val1 | val2

hoechenberger, 2020-07-30:
Turns out `find_bads_ecg` has **both**, a `measure` and a `method` kwarg which control different thing. Will therefore only slightly adjust the docstrings.

closed 2020-07-31 by larsoner