mne-tools/mne-python #8069: Crop-Method-Documentation inconsistent across Raw/Epochs/Evokeds

marsipu, 2020-07-29:
#### Describe the problem
The documentation of the method "crop" for Raw/Epochs/Evokeds is inconsistent:
For all Types, crop modifies the object in-place **and** returns the modified instance
Raw: In-Place-Modification is described in "Returns"
Epochs: In-Place-Modification is described in "Notes"
Evokeds: In-Place-Modification is **not** described

#### Describe your solution
Correction of the documentation

#### Additional context
If not too complicated, an "inplace"-argument would be helpful to avoid the additional "copy"-method

agramfort, 2020-07-29:
do you think you can give it a try to send the pull request with the fixes?

>

marsipu, 2020-07-29:
Yes, I'll try it on the weekend

hoechenberger, 2020-07-30:
> If not too complicated, an "inplace"-argument would be helpful to avoid the additional "copy"-method

I'd like that too but I don't think it's going to happen :) So please only try fixing the docstrings for now.
Thanks!

marsipu, 2020-07-31:
Ok

marsipu, 2020-07-31:
I tried to get GNU-Make to run under Windows or WSL2, but both didn't work yet for different reasons:

##Windows
After replacing `BUILD_DEV_HTML=1` with `setx BUILD_DEV_HTML 1` to make it run I ended up with a sphinx error:
 `Could not import extension sphinxcontrib.bibtex2`

##WSL2
OpenGL seems to be the problem for showing the documentation with sphinx.
`2020-07-31 18:53:37.619 (   1.543s) [        A54F0740]vtkXOpenGLRenderWindow.:251    ERR| vtkXOpenGLRenderWindow (0x7ffff0470240): Could not find a decent config
2020-07-31 18:53:37.621 (   1.545s) [        A54F0740]vtkXOpenGLRenderWindow.:468    ERR| vtkXOpenGLRenderWindow (0x7ffff0470240): Could not find a decent visual`

Has somebody solved these problems already or got another way of testing on Windows?

_caution: testing won't work on windows, see README_
(I saw this in Makefile, but didn't find anything in README, so I thought, maybe somebody found a solution)

marsipu, 2020-07-31:
I had some problems at the beginning, mostly because I use Windows and read the contribution-guide of the stable-version:) (missed everything about Windows and requirements_doc.txt:)

But one problem stayed:
I had to replace `BUILD_DEV_HTML=1` with `setx BUILD_DEV_HTML 1` in Makefile to make it run under Windows
(I don't know, if that's the best solution)

PR #8077 (2020-08-03): Fixing inconsistent documentation crop BaseEpochs, Evoked
#### Reference issue
Fixes #8069 .

#### What does this implement/fix?
Fixes the docstrings of BaseEpochs and Evoked, to make the documentation consistent and correct across BaseRaw/BaseEpochs/Evoked
closed 2020-08-03 by hoechenberger