mne-tools/mne-python #8071: Issues loading fieldtrip data with MNE python

acquayefrank, 2020-07-30:
Sorry, I did not know where else I could ask this:
I am having challenges loading fieldtrip data with MNE python. When I enter the code below

```
import mne
original_data = mne.io.read_raw_fif('data.fif', preload=False)
original_info = original_data.info
data_from_ft = mne.read_epochs_fieldtrip('data.mat', original_info)
```
I get this error:

`TypeError: only size-1 arrays can be converted to Python scalars `

larsoner, 2020-07-30:
Can you paste the full traceback? From just the last line it's hard to tell where the error comes from

drammock, 2020-07-30:
> Sorry, I did not know where else I could ask this

FYI, MNE-Python's website has a [how to get help page](https://mne.tools/stable/overview/get_help.html) that provides links to [our mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) and our [gitter channel](https://gitter.im/mne-tools/mne-python).

acquayefrank, 2020-07-30:
`<ipython-input-2-ff985abfd0ca>:1: RuntimeWarning: This filename (../data/ML_Subject08_P2_tsss_mc.fif) does not conform to MNE naming conventions. All raw files should end with raw.fif, raw_sss.fif, raw_tsss.fif, raw.fif.gz, raw_sss.fif.gz, raw_tsss.fif.gz or _meg.fif
  original_data = mne.io.read_raw_fif('../data/ML_Subject08_P2_tsss_mc.fif', preload=False)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-2-ff985abfd0ca> in <module>
      1 original_data = mne.io.read_raw_fif('../data/ML_Subject08_P2_tsss_mc.fif', preload=False)
      2 original_info = original_data.info
----> 3 data_from_ft = mne.read_epochs_fieldtrip('../data/dat_ft.mat', original_info)

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/fieldtrip/fieldtrip.py in read_epochs_fieldtrip(fname, info, data_name, trialinfo_column)
    118     _validate_ft_struct(ft_struct)
    119 
--> 120     info = _create_info(ft_struct, info)  # create info structure
    121     data = np.array(ft_struct['trial'])  # create the epochs data array
    122     events = _create_events(ft_struct, trialinfo_column)

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/fieldtrip/utils.py in _create_info(ft_struct, raw_info)
     72         info['sfreq'] = sfreq
     73         ch_idx = [info['ch_names'].index(ch) for ch in ch_names]
---> 74         pick_info(info, ch_idx, copy=False)
     75     else:
     76         montage = _create_montage(ft_struct)

<decorator-gen-8> in pick_info(info, sel, copy, verbose)

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/pick.py in pick_info(info, sel, copy, verbose)
    468     from .meas_info import _bad_chans_comp
    469 
--> 470     info._check_consistency()
    471     info = info.copy() if copy else info
    472     if sel is None:

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/meas_info.py in _check_consistency(self, prepend_error)
    730         for key in ('sfreq', 'highpass', 'lowpass'):
    731             if self.get(key) is not None:
--> 732                 self[key] = float(self[key])
    733 
    734         # Ensure info['chs'] has immutable entries (copies much faster)

TypeError: only size-1 arrays can be converted to Python scalars`

acquayefrank, 2020-08-03:
```
<ipython-input-6-ff985abfd0ca>:1: RuntimeWarning: This filename (../data/ML_Subject08_P2_tsss_mc.fif) does not conform to MNE naming conventions. All raw files should end with raw.fif, raw_sss.fif, raw_tsss.fif, raw.fif.gz, raw_sss.fif.gz, raw_tsss.fif.gz or _meg.fif
  original_data = mne.io.read_raw_fif('../data/ML_Subject08_P2_tsss_mc.fif', preload=False)
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/fieldtrip/utils.py in _set_sfreq(ft_struct)
    217     try:
--> 218         sfreq = ft_struct['fsample']
    219     except KeyError:

KeyError: 'fsample'

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
<ipython-input-6-ff985abfd0ca> in <module>
      1 original_data = mne.io.read_raw_fif('../data/ML_Subject08_P2_tsss_mc.fif', preload=False)
      2 original_info = original_data.info
----> 3 data_from_ft = mne.read_epochs_fieldtrip('../data/dat_ft.mat', original_info)

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/fieldtrip/fieldtrip.py in read_epochs_fieldtrip(fname, info, data_name, trialinfo_column)
    111     _validate_ft_struct(ft_struct)
    112 
--> 113     info = _create_info(ft_struct, info)  # create info structure
    114     data = np.array(ft_struct['trial'])  # create the epochs data array
    115     events = _create_events(ft_struct, trialinfo_column)

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/fieldtrip/utils.py in _create_info(ft_struct, raw_info)
     46         warn(NOINFO_WARNING)
     47 
---> 48     sfreq = _set_sfreq(ft_struct)
     49     ch_names = ft_struct['label']
     50     if raw_info:

~/miniconda3/envs/srtt/lib/python3.8/site-packages/mne/io/fieldtrip/utils.py in _set_sfreq(ft_struct)
    223             raise ValueError('No Source for sfreq found')
    224         else:
--> 225             t1, t2 = float(time[0]), float(time[1])
    226             sfreq = 1 / (t2 - t1)
    227     try:

TypeError: only size-1 arrays can be converted to Python scalars
```

larsoner, 2020-08-03:
after you hit the error, can you do something like:
```
>>> import pdb; pdb.pm()
(pdb) p ft_struct['time']
```
And let me know the output? It will tell us the actual size/shape of the array.

acquayefrank, 2020-08-07:
Hello @larsoner,

Sorry for the late reply. Had issues with my laptop.
```
[array([-0.3  , -0.299, -0.298, ...,  0.698,  0.699,  0.7  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ]), array([-0.5  , -0.499, -0.498, ...,  0.498,  0.499,  0.5  ])]
```
I have pasted the results above.

larsoner, 2020-08-07:
Hmmm, not actually sure how best to handle this case. It looks like your epochs in FieldTrip have different time spans, which MNE does not currently support.

agramfort, 2020-08-08:
I would say not supported

acquayefrank, 2020-08-09:
Thanks @larsoner

larsoner, 2020-09-11:
I think this gives an informative error in latest `master` from #7802

PR #8075 (2020-08-03): BUG: Better float checking for FT
Closes #8071.

@acquayefrank can you see if this helps with your error at all?
closed 2020-09-11 by larsoner