mne-tools/mne-python #8076: Difficulty loading edf files

beapdk, 2020-07-31:
Hi, 

Using the mne.io.read_raw_edf, i can't read the edf data file and instead get this error :
Extracting EDF parameters from /home/bea/Documents/Example-Analysis/Data/sub-TDCS1_ses-T1_task-Pre_eeg.edf...
EDF file detected
Setting channel info structure...
Creating raw.info structure...
<ipython-input-3-6f39a157b592>:1: RuntimeWarning: Physical range is not defined in following channels:
E1001
  Raw = mne.io.read_raw_edf('Data/sub-TDCS1_ses-T1_task-Pre_eeg.edf')
<RawEDF | sub-TDCS1_ses-T1_task-Pre_eeg.edf, 130 x 787000 (787.0 s), ~258 kB, data not loaded>

I would like to not how to correct this issue !

Thank you !

drammock, 2020-07-31:
The "data not loaded" message means the data are not *preloaded* into computer RAM; that is normal behavior when `preload=False` (which is the default). Either specify `preload=True` when reading the file, or use the `Raw.preload()` method after reading the file.

closed 2020-08-01 by agramfort

cbrnr, 2020-08-01:
@beapdk in case you were referring to the `RuntimeWarning`, this just means that the physical range for one channel is not properly defined in the EDF header, therefore the scaling might be off. It is not an error though, so the data can be loaded (but as @drammock pointed out you should specify `preload=True` to actually load the file into memory).