mne-tools/mne-python #8078: Epochs: Baseline-Default problems

marsipu, 2020-08-02:
#### Describe the bug
Their are two problems regarding the Baseline-Argument of the Epochs-Classes:

1. The default-value (None, 0) is not correctly formatted by sphinx.ext.autodoc and results in "baseline=None, 0, ...
2. Epochs-Array doesn't have the same default-value as the other Epochs-Classes (None instead of (None, 0).

[image]
[image]

larsoner, 2020-08-02:
For `EpochsArray` I think that choice is intentional. If you do `EpochsArray` we probably shouldn't modify the array by default. For `Epochs` probably you want a baseline. So maybe just a `.. note::` in the `EpochsArray` docs saying it's different from `Epochs` is a good choice.

For the docs of a tuple being broken, someone will have to look into it. I have a feeling https://github.com/numpy/numpydoc/pull/221 or some similar refactoring in numpydoc could be to blame. If you want to try `git bisect` on `Sphinx` and/or `numpydoc` feel free -- if you're not up for it let me know and I'll give it a shot.

larsoner, 2020-08-04:
Upstream bug filed https://github.com/sphinx-doc/sphinx/issues/8052

closed 2020-08-04 by larsoner

marsipu, 2020-08-05:
Thank you @larsoner, sorry, I somehow missed your response. Thanks for fixing it