mne-tools/mne-python #8080: Using raw.plot in JupyterLab or Colab

gurasog, 2020-08-03:
Is there any recommendation how to use MNE plot() functions such as raw.plot() in JupyterLab and Google Colab?

When i run these functions they aren't interactive. None of %matplotlib %matplotlib qt works in these environment...

So, is there any way to use these graphs interactively in these tools?

Thanks!

larsoner, 2020-08-03:
What do you get for `mne.sys_info()` or `mne sys_info` on those platforms? Maybe xvfb+Qt5Agg backend would work

drammock, 2020-08-03:
I am able to get interactive `raw.plot()` windows using `%matplotlib qt` or `%matplotlib qt5` in JupyterLab (it pops open a separate window, not an inline figure).

Google Colab is "not officially supported" at the moment, though mainly because of problems with 3D visualization.  For 2D visualization it is expected to work the same as Jupyter Notebooks.  I tried to get Colab 3D viz working in June, and kept hitting segfaults (there is some discussion of this on [the Gitter channel](https://gitter.im/mne-tools/mne-python), search for "colab").

larsoner, 2021-02-19:
Closing since it seems like it works at least in JupyterLab but let's reopen if it doesn't actually

GuillaumeFavelier, 2021-09-02:
I can give a try tomorrow. In any case, I'll update https://github.com/mne-tools/mne-python/issues/8704 with this. Thanks for the update!

GuillaumeFavelier, 2021-09-03:
I managed to reproduce the [minimal example](https://github.com/googlecolab/colab-cdn-widget-manager/blob/main/notebooks/ipympl.ipynb) using `matplotlib` :

[image]

<details>

```
!pip install -q ipympl
```

```
%matplotlib widget
from IPython.display import Javascript
import matplotlib.pyplot as plt
import numpy as np

# This enables custom widgets in the current cell's output.
# Currently this must be included in each cell using custom widgets but this
# should be a temporary restriction.
display(Javascript('''
  google.colab.widgets.installCustomManager('http://127.0.0.1:9897/manager.dev.js');
'''))

from mpl_toolkits.mplot3d import axes3d

fig = plt.figure()
ax = fig.add_subplot(111, projection='3d')

# Grab some test data.
X, Y, Z = axes3d.get_test_data(0.05)

# Plot a basic wireframe.
ax.plot_wireframe(X, Y, Z, rstride=10, cstride=10)

plt.show()
```

</details>

but not [the one](https://mybinder.org/v2/gh/martinRenou/ipycanvas/stable?urlpath=lab%2Ftree%2Fexamples) using `ipycanvas` (`clock.ipynb`):

<details>

```
!pip install ipycanvas
```

```
from IPython.display import Javascript

display(Javascript('''
  google.colab.widgets.installCustomManager('http://127.0.0.1:9897/manager.dev.js');
'''))

from ipycanvas import Canvas, hold_canvas
import numpy as np

CLOCK_RADIUS = 100

canvas = Canvas(width=CLOCK_RADIUS*2.5, height=CLOCK_RADIUS*2.5)
canvas.translate(CLOCK_RADIUS * 1.2, CLOCK_RADIUS * 1.2)
display(canvas)

def clear_drawing():
    canvas.clear_rect(-CLOCK_RADIUS * 1.2, -CLOCK_RADIUS * 1.2, canvas.width, canvas.height)

def minutes_vec(minutes):
    a = minutes * np.pi/30
    return [np.sin(a), -np.cos(a)]

def draw_dial():
    ht = 10
    mt = 6
    ho = 20
    mo = 10

    canvas.text_align = 'center'
    canvas.text_baseline = 'middle'

    canvas.line_width = 2
    canvas.stroke_circle(0, 0, CLOCK_RADIUS)

    for m in range(60):
        a = m * np.pi/30
        x, y = np.sin(a), -np.cos(a)

        canvas.line_width = 1
        if m % 5 == 0:
            canvas.stroke_line(x * (CLOCK_RADIUS - ht), y * (CLOCK_RADIUS - ht), x * CLOCK_RADIUS, y * CLOCK_RADIUS)
            canvas.font = '12px serif'
            canvas.stroke_text(str(m), x * (CLOCK_RADIUS + mo), y * (CLOCK_RADIUS + mo))
            canvas.font = '16px serif'
            canvas.stroke_text(str(m//5 if m > 0 else 12), x * (CLOCK_RADIUS - ho), y * (CLOCK_RADIUS - ho))
        else:
            canvas.stroke_line(x * (CLOCK_RADIUS - mt), y * (CLOCK_RADIUS - mt), x * CLOCK_RADIUS, y * CLOCK_RADIUS)

def draw_hands(minutes):
    ms = 35
    hs = 50

    hrs = minutes // 60
    mins = minutes % 60

    mv = minutes_vec(mins)
    hv = minutes_vec(hrs * 5 + (mins / 12))

    canvas.line_width = 5
    canvas.stroke_line(0, 0, mv[0] * (CLOCK_RADIUS - ms), mv[1] * (CLOCK_RADIUS - ms))
    canvas.stroke_line(0, 0, hv[0] * (CLOCK_RADIUS - hs), hv[1] * (CLOCK_RADIUS - hs))

def draw_clock(hours, minutes):
    with hold_canvas(canvas):
        clear_drawing()
        draw_dial()
        draw_hands((hours % 12) * 60 + minutes)
```

</details>

So the [basic example](https://github.com/Kitware/ipyvtklink#python-vtk) using `ipyvtk` did not work either:

<details>

```
!pip install vtk
!pip install ipyvtklink
```

```
from IPython.display import Javascript
import vtk
from ipyvtklink.viewer import ViewInteractiveWidget

# This enables custom widgets in the current cell's output.
# Currently this must be included in each cell using custom widgets but this
# should be a temporary restriction.
display(Javascript('''
  google.colab.widgets.installCustomManager('http://127.0.0.1:9897/manager.dev.js');
'''))

# Create some data
cylinder = vtk.vtkCylinderSource()
cylinder.SetResolution(8)
mapper = vtk.vtkPolyDataMapper()
mapper.SetInputConnection(cylinder.GetOutputPort())
actor = vtk.vtkActor()
actor.SetMapper(mapper)

# Set up render window
ren = vtk.vtkRenderer()
ren_win = vtk.vtkRenderWindow()
ren_win.SetOffScreenRendering(1)
ren_win.SetSize(600, 600)
ren_win.AddRenderer(ren)
iren = vtk.vtkRenderWindowInteractor()
iren.SetRenderWindow(ren_win)
style = vtk.vtkInteractorStyleTrackballCamera()
iren.SetInteractorStyle(style)

# Add actor to scene
ren.AddActor(actor)
ren.ResetCamera()

# Display
ViewInteractiveWidget(ren_win)
```

</details>

They both do not give any output

larsoner, 2021-09-03:
In the issue linked above they state:

> Please feel free to open specific bugs in https://github.com/googlecolab/colab-cdn-widget-manager/issues.

Can you see if this has been reported in https://github.com/googlecolab/colab-cdn-widget-manager/issues and if not, add this information there? Seems like useful debugging for them!

GuillaumeFavelier, 2021-09-03:
The related issue is opened in https://github.com/googlecolab/colab-cdn-widget-manager/issues/8 !

larsoner, 2021-09-08:
Sounds like they've fixed it: https://github.com/googlecolab/colab-cdn-widget-manager/issues/8#issuecomment-915454227

larsoner, 2022-03-09:
I'll close this as upstream bug(s), but people feel free to comment if it's fixed yet (or not)!

closed 2022-03-09 by larsoner