mne-tools/mne-python #8085: Getting Error: "IndexError: index 0 is out of bounds for axis 0 with size 0"

balandongiv, 2020-08-04:
#### Describe the bug
〈Replace this text with a clear and concise description of the bug.〉

While executing `ica.plot_properties(raw, picks=ecg_indices)`, the compiler return an error

`IndexError: index 0 is out of bounds for axis 0 with size 0`

The full error trail:

```
    ica.plot_properties(raw, picks=ecg_indices)
  File "lib\site-packages\mne\preprocessing\ica.py", line 1746, in plot_properties
    figsize=figsize, show=show, reject=reject)
  File "\lib\site-packages\mne\viz\ica.py", line 427, in plot_ica_properties
    psds, freqs = psd_multitaper(epochs_src, picks=picks, **psd_args)
  File "<decorator-gen-61>", line 21, in psd_multitaper
  File "\lib\site-packages\mne\time_frequency\psd.py", line 334, in psd_multitaper
    n_jobs=n_jobs, verbose=verbose)
  File "<decorator-gen-48>", line 21, in psd_array_multitaper
  File "\lib\site-packages\mne\time_frequency\multitaper.py", line 422, in psd_array_multitaper
    n_times, sfreq, bandwidth, low_bias, adaptive)
  File "<decorator-gen-47>", line 21, in _compute_mt_params
  File "\lib\site-packages\mne\time_frequency\multitaper.py", line 350, in _compute_mt_params
    interp_from=interp_from)
  File "lib\site-packages\mne\time_frequency\multitaper.py", line 95, in dpss_windows
    dpss = _get_dpss()(N, half_nbw, Kmax)
  File "\lib\site-packages\scipy\signal\windows\windows.py", line 1938, in dpss
    if w[w * w > thresh][0] < 0:
IndexError: index 0 is out of bounds for axis 0 with size 0
```

#### Steps to reproduce
    filename = r'S17_3.mff' # can be downloaded via the link: https://drive.google.com/drive/folders/14R_2A-ceqOFvRkDZH6Skm4CqF5B-K7Co?usp=sharing
    raw = mne.io.read_raw_egi(filename, preload=True)
    montage = mne.channels.make_standard_montage('GSN-HydroCel-129')
    montage.ch_names[-1] = 'E129'
    raw.set_montage(montage, match_case=False)
    raw.filter(1., 40.)
    raw.crop(0, 595)  # crop huge artifact at the end
    filt_raw = raw.copy()

    ica = ICA(n_components=30, random_state=97)
    ica.fit(filt_raw)
    ica.exclude = []

    # find which ICs match the ECG pattern
    ecg_indices, ecg_scores = ica.find_bads_ecg(raw)
    ica.exclude = ecg_indices

    # barplot of ICA component "ECG match" scores
    ica.plot_scores(ecg_scores)

    # plot diagnostics
    ica.plot_properties(raw, picks=ecg_indices)

    # plot ICs applied to raw data, with ECG matches highlighted
    ica.plot_sources(raw)

Additional information

System:

Windows 10
anaconda3
mne-0.21.dev0
Python 3.7
[Link to download S17_3.mff ](https://drive.google.com/drive/folders/14R_2A-ceqOFvRkDZH6Skm4CqF5B-K7Co?usp=sharing)

larsoner, 2020-08-04:
I can't replicate on Linux, it's possible that this is a windows issue. Before I boot a VM to look into it, when you hit the error can you try:
```
>>> import pdb; pdb.pm()
(Pdb) p (NW, Kmax, sym, norm, M, thresh, np.abs(m).max())
...
```
and let us know the output? It can help with debugging

balandongiv, 2020-08-04:
Thanks for the response @larsoner .  Im using PyCharm and I am not familiar with `pdb` for troubleshooting. May I know other approach to capture specific output that you have in mind?

larsoner, 2020-08-04:
Can you run Python from a standard terminal instead? Copy-pasting your code (with an `import mne` at the top) into some file then do:
```
python -i whatever.py
```
and you should end up dropped to a prompt where you can `import pdb; pdb.pm()`. But really I'm surprised Pycharm doesn't give you a way to interact with the Python terminal once a script completes. It's standard Python behavior if you pass `python -i` or use "interactive mode".

It's also likely that pycharm has some built-in debugging capability where you can have it drop to a prompt if it hits an error.

balandongiv, 2020-08-05:
Thanks for the clear instruction @larsoner . Surprisingly, the plot was rendered without any hiccup when executing the commands using "interactive mode". However, there is a catch whereby the figure was plotted properly only during the first execution (i.e.,  execute only the `ica.plot_properties(raw, picks=ecg_indices`)). But, after I close the pop figure and execute the line` ica.plot_properties(raw, picks=ecg_indices)` , the same error  `"IndexError: index 0 is out of bounds for axis 0 with size 0"` was shot.
The same error also was given if executing the line` ica.plot_properties(raw, picks=ecg_indices)` while the first figure is still open.

larsoner, 2020-08-10:
Hopefully this was fixed by #8103, let's reopen if not

closed 2020-08-10 by larsoner

behinger, 2020-12-03:
I still have the same problem with todays 0.22dev version, also on windows 10. If I run it through debugger it works. Somtimes it works one time and not the second

If I run your post-mortem debugging thingy I get this:
```(Pdb) p (NW, Kmax, sym, norm, M, thresh, np.abs(M).max())
 (4.0, 8, True, 2, 2048, 0.00048828125, 2048)```

(note that I changed the np.abs(m) to np.abs(M) hope that was intended)

larsoner, 2020-12-04:
@behinger do you have NumPy 1.19.4? I wonder if it's this Windows bug:

https://developercommunity2.visualstudio.com/t/fmod-after-an-update-to-windows-2004-is-causing-a/1207405
https://github.com/numpy/numpy/issues/16744

For now the workaround is to install NumPy 1.19.3 on Windows.

behinger, 2020-12-04:
nice! yes indeed I had Numpy 1.19.2 installed. With 1.19.3 it works now.

Thanks alot! it might also explain random SVD errors I got :)

LiamMBooth, 2020-12-10:
Is there a specific Scipy library we should maybe be using too? 

I've got a similar error when trying to open multiple figures from my epochs object and have tried installing my Numpy to version 1.19.3 and have MNE 0.21.2 installed. I'm using a Windows machine too.