mne-tools/mne-python #8087: Error while reading edf file

LeyloJohnny, 2020-08-04:
EDF file was detected but couldn't be read.

Code:

```
import mne
import os 

datapath = '(...anonymous due to data protection)'
file2read = os.path.join(datapath,'testfile.edf')

raw = mne.io.read_raw_edf(file2read, preload = True)

```

Error:

```

Extracting EDF parameters from (...anonymous due to data protection)
EDF file detected
Traceback (most recent call last):
  File "..Skripte\import mne.py", line 7, in <module>
    raw = mne.io.read_raw_edf(file2read, preload = True)
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py", line 1181, in read_raw_edf
    return RawEDF(input_fname=input_fname, eog=eog, misc=misc,
  File "<decorator-gen-171>", line 21, in __init__
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py", line 113, in __init__
    info, edf_info, orig_units = _get_info(input_fname,
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py", line 377, in _get_info
    edf_info, orig_units = _read_header(fname, exclude)
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py", line 364, in _read_header
    return _read_edf_header(fname, exclude)
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\edf\edf.py", line 532, in _read_edf_header
    with open(fname, 'rb') as fid:
FileNotFoundError: [Errno 2] No such file or directory: '(...anonymous due to data protection)'

```

My system info:

```

Platform:      Windows-10-10.0.18362-SP0
Python:        3.8.3 (default, Jul  2 2020, 17:30:36) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\ProgramData\Anaconda3\python.exe
CPU:           Intel64 Family 6 Model 94 Stepping 3, GenuineIntel: 8 cores
Memory:        7.9 GB

mne:           0.21.dev0
numpy:         1.18.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.0
matplotlib:    3.2.2 {backend=Qt5Agg}

sklearn:       0.23.1
numba:         0.50.1
nibabel:       Not found
cupy:          Not found
pandas:        1.0.5
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

```

larsoner, 2020-08-04:
The detection is just based on the file extension. You need to provide a file2read that returns True for os.path.isfile(file2read) -- is that the case for your variable? Those `...` in the filename might need escapes for example.

LeyloJohnny, 2020-08-04:
Thank You for the fast response!
I didn't get true for os.path.isfile(file2read).
I will check for my datapath and file...

closed 2020-08-04 by LeyloJohnny