mne-tools/mne-python #8088: ENH: Improve our sourcerank score

larsoner, 2020-08-04:
See some entries missing in

https://libraries.io/pypi/mne/sourcerank

compared to 

https://libraries.io/pypi/nibabel/sourcerank

For 0.21 we should fix the easy missing ones. Something trivial for someone to do as a Sunday hack...

drammock, 2020-08-04:
How does one even fix such things?  When I'm logged in, I don't see any "add" or "edit" links...  MNE-Python master doesn't even seem to be on the site? but some random fork of MNE is: https://libraries.io/github/Sinaxist/mne-python

larsoner, 2020-08-04:
I poked around and "submitted a suggestion" to link it to `mne-tools/mne-python`

larsoner, 2020-09-09:
Closing as I guess this is "good enough" for now

drammock, 2020-09-21:
I think I figured out why this is happening: our [PyPI json](https://pypi.org/pypi/mne/json) is missing the key `['project_urls']['Source']`.  Here is [an example](https://pypi.org/pypi/doit/json) of a project that has the key defined.  I figured this out from [this issue](https://github.com/librariesio/libraries.io/issues/2121), and I think the solution is to add [something like this](https://github.com/pydoit/doit/blob/d1a1b7b3abc7641d977d3b78b580d97aea4e27ea/setup.py#L56-L59) to our `setup.py`.  @larsoner WDYT?

larsoner, 2020-09-22:
Yes please add it @drammock . You can probably check by doing `python setup.py sdist bdist_wheel` then looking at the `dist/` directory, the JSON should be in there somewhere. Once it works we can merge to `master` and backport.

drammock, 2020-09-23:
hmm, the JSON is not in `dist` anywhere, but the new URLs do show up in the `mne.egg-info/PKG-INFO` inside the `tar.gz` file, and also in `python setup.py sdist bdist_wheel/METADATA` inside the wheel. I'll open a PR to iterate further.

PR #8297 (2020-09-23): Sourcerank
closes #8088
closed 2020-09-23 by drammock