mne-tools/mne-python #8094: Go to Plot Properties via clicking the IC component within the Component Plot

balandongiv, 2020-08-05:
Within the `ica.plot_sources()`, one can view to the topographical 2-D scalp maps for each IC component by clicking the `Y-axis`. Whereas, as per my knowledge, the only way to generate the diagnostics of each IC is by using `ica.plot_properties(raw, picks=[3,10,15])`. 
Is it possible to view the diagnostics of each IC simply by clicking the` Y-axis` from the `ica.plot_sources()`?

I found it quite troublesome to key in the picks index to generate diagnostics for different IC

larsoner, 2020-08-05:
@cbrnr perhaps one for you?

cbrnr, 2020-08-05:
Would this be relevant for #7751? Or at least after #7751 has been merged?

hoechenberger, 2020-08-05:
I would love to see this implemented too :)

drammock, 2020-08-05:
I think it's appropriate to add this to #7751 (it might even sneak into #7955).  My only objection is that whatever we pop open when clicking on channel/component names should be something that plots *quickly*.  I haven't timed it, but `ica.plot_properties` is going to be slower (in addition to the scalp field topography, it makes 3 other subplots including one that estimates the spectrum).  Does this change anyone's mind?

drammock, 2020-08-05:
I guess another consideration is similarity of behavior across instance types.  As I understand it, the unification plan is that **left-clicking** on channel names should always mark bad channels / mark components to exclude, and **right-clicking** would do this:

- raw: show channel location on head (currently, nothing happens)
- epochs: show `epochs.plot_image()` for that channel
- ICA: show scalp field topography (current behavior); show `ica.plot_properties()` for that component (proposed here)

It's not clear to me that one or the other of the ICA right-click behaviors is "more consistent" or "more in line with user expectations" here,  so I guess I'm happy to go with what users say they want.  Does anyone disagree?

cbrnr, 2020-08-05:
I don't disagree - we should strive for consistency, but in this case it is not clear what consistent means. I guess whatever is the most useful behavior for right-clicking should be implemented. Personally, I'd be fine with keeping the topomap, but if others think that IC properties are more useful I'm also OK with that.

hoechenberger, 2020-08-05:
@drammock 
> I haven't timed it, but `ica.plot_properties` is going to be slower (in addition to the scalp field topography, it makes 3 other subplots including one that estimates the spectrum). Does this change anyone's mind?

For me personally having to wait a little would not be a deal breaker when it comes to manual IC inspection. I think key to a good UX in spite of a little waiting time is clearly informing the user about what is going on. I'm thinking opening the popup immediately upon clicking, and displaying `Processing …` or something similar either in the center of the figure or in the subplots individually until they've finished rendering one by one.

drammock, 2020-08-05:
> I'm thinking opening the popup immediately upon clicking, and displaying `Processing …` or something similar either in the center of the figure or in the subplots individually until they've finished rendering one by one.

I won't promise to do that (at least not as part of #7955).  Could be added later if it seems like it's needed.  But I'll plan to hook up the right-click-on-component-name to `ica.plot_properties()`

hoechenberger, 2020-08-05:
> I won't promise to do that (at least not as part of #7955).  Could be added later if it seems like it's needed.

Yes, absolutely. Let's first see how it feels without any progress indicator. It can always be added later on if need be.

JiangTao100, 2020-08-06:
I have the same problem in picking index to generate diagnostics for different IC by  ica. plot_ properties( ).When  I run it , there will
be an error displaying  ‘  IndexError : index 161 is out of bounds for axis 0 with size 161’.