mne-tools/mne-python #8096: Progressbar not visible in Spyder

mmagnuski, 2020-08-05:
Performing a cluster based test (unfortunatelly one that takes a lot of time) I no longer see the progressbar in Spyder:
```
stat_fun(H1): min=-5.699470 max=3.482243
Running initial clustering
Found 17 clusters
Permuting 9999 times...
HBox(children=(FloatProgress(value=0.0, max=9999.0), HTML(value='')))
```
this is most likely a result of using auto mode of `tqdm` progressbar, which seems to be unable to tell whether it is in a notebook or ipython console. It would be nice to have some option to control this behavior (`progressbar='text'` maybe?) as now everyone who works in ipython/Spyder will likely have no progressbar, which for long running tasks is a bit inconvenient.

larsoner, 2020-08-05:
This sounds more like an upstream bug with tqdm. Can you look into fixing it there (if it's not already fixed; try using tqdm locally as IIRC we prefer an installed system version when available and new enough)?

mmagnuski, 2020-08-05:
did github automatically self-assigned me? weird, I just clicked "submit". Anyway - I can do it.

@larsoner 
Unfortunatelly from what I read on multiple occasions it is not possible to distinguish whether the code is running from notebook or ipython (or Spyder). :(

larsoner, 2020-08-05:
Perhaps they should support magic `%tqdm text` for example to set within a given session which auto backend will be selected. Might be worth proposing and implementing that at their end. If they reject it, we can add some MNE config value that would triage the class used, but it makes more sense for them to implement it if they're up for it.

mmagnuski, 2020-08-05:
@larsoner Good idea, I'll suggest that there. Although the response I imagine might be "just use `tqdm.tqdm` instead of auto or `tqdm.notebook`", but I'll try. :)

larsoner, 2020-08-05:
> Although the response I imagine might be "just use tqdm.tqdm instead of auto or tqdm.notebook", but I'll try. :)

Yes if they do this we can support it locally. But they will be forcing every project that uses `tqdm` to reimplement this same triaging mechanism. So hopefully they see the value in it :)

larsoner, 2020-08-24:
@mmagnuski any progress at the `tqdm` end? We're recommending Spyder as the IDE in our release, so if they're not fixing it, we should.

PR #8177 (2020-09-07): MRG, ENH: Allow setting tqdm backend
Closes #8096

@mmagnuski want to give it a try?
closed 2020-09-07 by hoechenberger