mne-tools/mne-python #8101: ica.plot_properties error bug

neuroteams, 2020-08-06:
#### Describe the bug
ica.plot_properties doesnt work for raw annotated with bad segments

#### Steps to reproduce
import os
import mne
from mne.preprocessing import (ICA, create_eog_epochs, create_ecg_epochs,
                               corrmap)

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw.crop(tmax=60.)

a = mne.Annotations(onset=[3, 5, 7],
                           duration=[1, 0.5, 0.25],
                           description=['BAD', 'BAD', 'BAD'])
raw.set_annotations(a)

ica = mne.preprocessing.ICA()#max_pca_components=29)
ica.fit(raw)

ica.plot_properties(raw, picks=[0, 1])

#### Expected results
picks with component properties

#### Actual results
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-7-dae74f67a92b> in <module>
----> 1 ica.plot_properties(raw, picks=[0, 1])

/usr/local/anaconda3/lib/python3.7/site-packages/mne/preprocessing/ica.py in plot_properties(self, inst, picks, axes, dB, plot_std, topomap_args, image_args, psd_args, figsize, show, reject)
   1744                                    topomap_args=topomap_args,
   1745                                    image_args=image_args, psd_args=psd_args,
-> 1746                                    figsize=figsize, show=show, reject=reject)
   1747 
   1748     @copy_function_doc_to_method_doc(plot_ica_sources)

/usr/local/anaconda3/lib/python3.7/site-packages/mne/viz/ica.py in plot_ica_properties(ica, inst, picks, axes, dB, plot_std, topomap_args, image_args, psd_args, figsize, show, reject)
    404 
    405     epochs_src = ica.get_sources(inst_rejected)
--> 406     data = epochs_src.get_data()
    407 
    408     ica_data = np.swapaxes(data[:, picks, :], 0, 1)

/usr/local/anaconda3/lib/python3.7/site-packages/mne/epochs.py in get_data(self, picks, item)
   1440             A view on epochs data.
   1441         """
-> 1442         return self._get_data(picks=picks, item=item)
   1443 
   1444     @property

<decorator-gen-188> in _get_data(self, out, picks, item, verbose)

/usr/local/anaconda3/lib/python3.7/site-packages/mne/epochs.py in _get_data(self, out, picks, item, verbose)
   1367                     else:
   1368                         epoch_noproj = None
-> 1369                         epoch = self._data[idx]
   1370                 else:  # from disk
   1371                     epoch_noproj = self._get_epoch_from_raw(idx)

IndexError: index 27 is out of bounds for axis 0 with size 27〉

#### Additional information
Platform:      Linux-4.18.0-25-generic-x86_64-with-debian-buster-sid
Python:        3.7.7 (default, Mar 26 2020, 15:48:22)  [GCC 7.3.0]
Executable:    /usr/local/anaconda3/bin/python
CPU:           x86_64: 8 cores
Memory:        31.3 GB

mne:           0.21.dev0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       Not found
cupy:          Not found
pandas:        1.0.3
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

JiangTao100, 2020-08-06:
I have had the same problem 2 days ago. However, when marked no 'Bad_' ,it worked well. SO WEIRD!!

JiangTao100, 2020-08-07:
Successfully solve it .THANKS! @yh-luo

agramfort, 2020-08-07:
this is a bug that should be fixed.

PR welcome

>

neuroteams, 2020-08-09:
For me the WIP FIX Check reject_by_annotation in ica.plot_properties #8103
does'n help at all, I've got the same error:

/usr/local/anaconda3/lib/python3.7/site-packages/mne/viz/ica.py in plot_ica_properties(ica, inst, picks, axes, dB, plot_std, topomap_args, image_args, psd_args, figsize, show, reject)
    404     if getattr(ica, 'reject_by_annotation', False):
    405         inst_rejected.drop_bad()
--> 406     epochs_src = ica.get_sources(inst_rejected)
    407     data = epochs_src.get_data()
    408 

/usr/local/anaconda3/lib/python3.7/site-packages/mne/epochs.py in get_data(self, picks, item)
   1440             A view on epochs data.
   1441         """
-> 1442         return self._get_data(picks=picks, item=item)
   1443 
   1444     @property

<decorator-gen-188> in _get_data(self, out, picks, item, verbose)

/usr/local/anaconda3/lib/python3.7/site-packages/mne/epochs.py in _get_data(self, out, picks, item, verbose)
   1367                     else:
   1368                         epoch_noproj = None
-> 1369                         epoch = self._data[idx]
   1370                 else:  # from disk
   1371                     epoch_noproj = self._get_epoch_from_raw(idx)

IndexError: index 27 is out of bounds for axis 0 with size 27

yh-luo, 2020-08-09:
Hi @neuroteams , it looks like you used previous codes and not the latest, can you try the current PR?

neuroteams, 2020-08-10:
> Hi @neuroteams , it looks like you used previous codes and not the latest, can you try the current PR?

I use 0.21.dev0 , which version is current PR?

PR #8103 (2020-08-10): MRG, FIX: Check reject_by_annotation in ica.plot_properties
Fix #8101 .

Because `reject_by_annotation` in `ica.fit()` defaults to `True`, reconstruction of source in `plot_properties` should take care of that or the error occurs due to the incorrect number of epochs.

**TODO**
- _Fix `ica.plot_properties`_
- _Fix `make_fixed_length_epochs`_
- _Update documentation of `reject_by_annotation` using docdict_
- _Update what's new_

### Before this PR

After adding annotations
```python
---------------------------------------------------------------------------
IndexError                            [... cut]
closed 2020-08-10 by larsoner