mne-tools/mne-python #8106: How to epoch by using events_from_annotations ?

JiangTao100, 2020-08-07:
I‘ve learned that .CNT profiles store events in  raw. annotations . But when I wanna use the function 'mne. events_from_annotations' to epoch ,it displays 'TypeError: events should be a NumPy array of integers, got <class 'tuple'>' compared to my present tuple form.
Blow is my code:
#######
import os
import numpy as np
import mne
from mne.datasets import sample
import matplotlib.pyplot as plt
from mpl_toolkits.mplot3d import Axes3D
from mne.preprocessing import (ICA, create_eog_epochs, create_ecg_epochs,
                               corrmap)
fname='xxxxxx.cnt'
raw=mne.io.read_raw_cnt(fname)

event_raw=raw.copy()

a=mne.Annotations(onset=[2.1,4.0,53.5,54.0,59.0,60.8,156.4,168.2,208.6,209.2,240.4,250.4,290.5,324.0],duration=[0.2,0.2,0.3,0.2,1.0,0.9,0.2,1.8,0.4,0.1,0.2,0.2,2.5,3.5],description=['BAD'])
plus_anno=a+raw.annotations
event_raw.set_annotations(plus_anno)
event_dict=mne.events_from_annotations(event_raw)
epochs = mne.Epochs(event_raw,event_array
                    ,tmin=-0.3, tmax=0.7,
                    preload=True)
#####################
error displayed:
runcell('epoch', 'D:/python project/mypython/practice.py')
Used Annotations descriptions: ['11', '111', '112', '12', '121', '122']
Traceback (most recent call last):

  File "D:\python project\mypython\practice.py", line 82, in <module>
    preload=True)

  File "<decorator-gen-184>", line 21, in __init__

  File "D:\softinstall\Anaconda3\lib\site-packages\mne\epochs.py", line 2095, in __init__
    event_repeated=event_repeated, verbose=verbose)

  File "<decorator-gen-175>", line 21, in __init__

  File "D:\softinstall\Anaconda3\lib\site-packages\mne\epochs.py", line 399, in __init__
    'got {}'.format(events_type))

TypeError: events should be a NumPy array of integers, got <class 'type'>
########
Sorry for my CNT profile is too big to upload.

cbrnr, 2020-08-07:
`mne.events_from_annotations` returns a tuple of the actual events (NumPy array) and a description dict. Use `event_dict[0]` to access the event array.

closed 2020-08-07 by drammock

mokennen, 2022-12-29:
I'm working on. EDF file but when I try to epoch the data, get Type "Error: events should be a NumPy array of integers, got <class 'tuple'>" error. how can I  bypass this problem
with regerads
​

agramfort, 2022-12-29:
Events_from_annotations returns 2 things events and event_id 
Each one is one different parameter in Epochs

drammock, 2022-12-29:
@mokennen please use our user forum, https://mne.discourse.group for usage questions. Do not post comments on already-closed GitHub issue threads. 

Note also that @cbrnr already gave the likely solution to your problem in the comment immediately above yours:

> `mne.events_from_annotations` returns a tuple of the actual events (NumPy array) and a description dict.