mne-tools/mne-python #8110: In the code for recalibrating the sensor geometric parameters of the SSS method,why to calculate the "vector rotation matrix" between third column of the "info['chs']['loc']" rotation matrix and the third column of the "fine_cal['locs']" rotation matrix.

INNERMONGOLIAszm, 2020-08-09:
In the code for recalibrating the sensor geometric parameters of the SSS method,why to calculate the "vector rotation matrix" between third column of the "info['chs']['loc']" rotation matrix and the third column of the "fine_cal['locs']" rotation matrix.

drammock, 2020-08-09:
Normally, questions about algorithms should be directed to the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python).  But since you're asking about specific lines of code, it might be easier to work out the answer here on GitHub.  Could you follow [these instructions](https://docs.github.com/en/github/managing-your-work-on-github/creating-a-permanent-link-to-a-code-snippet) to link to the specific line(s) of code you're asking about?

INNERMONGOLIAszm, 2020-08-09:
> Normally, questions about algorithms should be directed to the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter [...]

https://github.com/mne-tools/mne-python/blob/421cb1bc2f432f8eb93a49a3c38be1ac48838497/mne/preprocessing/maxwell.py#L1682-L1696
        # Some .dat files might only rotate EZ, so we must check first that
        # EX and EY are orthogonal to EZ. If not, we find the rotation between
        # the original and fine-cal ez, and rotate EX and EY accordingly:
        ch_coil_rot = _loc_to_coil_trans(info_ch['loc'])[:3, :3]
        cal_loc = fine_cal['locs'][ci].copy()
        cal_coil_rot = _loc_to_coil_trans(cal_loc)[:3, :3]
        if np.max([np.abs(np.dot(cal_coil_rot[:, ii], cal_coil_rot[:, 2]))
                   for ii in range(2)]) > 1e-6:  # X or Y not orthogonal
            if not adjust_logged:
                logger.info('        Adjusting non-orthogonal EX and EY')
                adjust_logged = True
            # find the rotation matrix that goes from one to the other
            this_trans = _find_vector_rotation(ch_coil_rot[:, 2],
                                               cal_coil_rot[:, 2])
            cal_loc[3:] = np.dot(this_trans, ch_coil_rot).T.ravel()

The comment reads“Some .dat files might only rotate EZ”，I want to know what EZ is and why it's OK to just rotate it .Why not rotate EY and EX？

agramfort, 2020-08-10:
@larsoner is the person who can answer you.

>

larsoner, 2020-08-10:
IIRC the official fine cal binaries provided by Neuromag/MEGIN/Elekta only do "1D" fine cal, i.e., they only rotate EZ. It is possible to compute and use "3D" fine cal that tunes/modifies all three. But I think EZ is the most important one to correct, and the benefit from 3D is more subtle compared to no-fine-cal vs 1D (standard) fine cal.

closed 2020-08-10 by larsoner