mne-tools/mne-python #8114: channel rereference destroy future ICA

neuroteams, 2020-08-11:
#### Describe the bug
When I add the channel rereferense to the pipeline, I get the wrong ICA componets

#### Steps to reproduce
ref_channels=['T7', 'T8']
raw, _ = mne.set_eeg_reference(raw, ref_channels)

raw.filter(l_freq=1, h_freq=30)
montage_kind = "standard_1005"
montage = mne.channels.make_standard_montage(montage_kind)
raw.set_montage(montage, match_case=False)
ica = mne.preprocessing.ICA()
ica.fit(raw)
ica.plot_components()

#### Expected results
this image got with
raw.set_eeg_reference()
[image]

#### Actual results
this image got with
ref_channels=['T7', 'T8']
raw.set_eeg_reference(ref_channels)
[image]

#### Additional information
Platform:      Linux-4.18.0-25-generic-x86_64-with-debian-buster-sid
Python:        3.7.7 (default, Mar 26 2020, 15:48:22)  [GCC 7.3.0]
Executable:    /usr/local/anaconda3/bin/python
CPU:           x86_64: 8 cores
Memory:        31.3 GB

mne:           0.21.dev0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       Not found
cupy:          Not found
pandas:        1.0.3
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

cbrnr, 2020-08-11:
Indeed this looks weird and in theory ICA should be insensitive to the choice of reference. I would try two things:

1. Use a different algorithm; I assume you used the default `fastica`, which is not known to extract dipolar sources as well as other algorithms such as Infomax. I'd try `infomax` with `fit_params={'extended': True}` to see how the reference affects results.
2. Maybe the T7/T8 reference changes the rank of the data somehow (in theory it shouldn't), so you could try to set `max_pca_components` to a value one or two less than the number of channels you have.

hoechenberger, 2020-08-11:
> 2\. Maybe the T7/T8 reference changes the rank of the data somehow (in theory it shouldn't),

I would have expected it to reduce the rank by 2, but there's a big change that **my** theory is off 😅

hoechenberger, 2020-08-19:
Friendly ping @neuroteams, did @cbrnr's suggestion help you? If you're stuck, you may also share a problematic file with me privately so I can have a look: richard.hoechenberger@gmail.com

hoechenberger, 2020-10-02:
I'm closing this due to inactivity. @neuroteams feel free to re-open if this issue has not been resolved.

closed 2020-10-02 by hoechenberger