mne-tools/mne-python #8119: Hash mismatch for downloaded file C:\Users\BCI-13\mne_data\MNE-sample-data-processed.tar.gz.part

iuype, 2020-08-14:
I run the codes below.

`
import mne
from mne import io
from mne.datasets import sample
data_path = sample.data_path()
`
But I get an error.

`
Using default location ~/mne_data for sample...
Downloading archive MNE-sample-data-processed.tar.gz to C:\Users\BCI-13\mne_data
Downloading https://files.osf.io/v1/resources/rxvq7/providers/osfstorage/59c0e26f9ad5a1025c4ab159?action=download&direct&version=5 (1.54 GB)
[........................................] 100.00% ( 1.54 GB,  18.5 MB/s) \    
Verifying hash fc2d5b9eb0a144b1d6ba84dc3b983602.
Error while fetching file https://files.osf.io/v1/resources/rxvq7/providers/osfstorage/59c0e26f9ad5a1025c4ab159?action=download&direct&version=5. Dataset fetching aborted.
Traceback (most recent call last):

  File "C:\Users\BCI-13\Downloads\arl-eegmodels-master\ERP.py", line 91, in <module>
    data_path = sample.data_path()

  File "<string>", line 2, in data_path

  File "D:\Application\anaconda3\envs\py36-tf\lib\site-packages\mne\utils.py", line 952, in verbose
    return function(*args, **kwargs)

  File "D:\Application\anaconda3\envs\py36-tf\lib\site-packages\mne\datasets\sample\sample.py", line 21, in data_path
    download=download)

  File "D:\Application\anaconda3\envs\py36-tf\lib\site-packages\mne\datasets\utils.py", line 389, in _data_path
    remove_archive, full = _download(path, u, an, h)

  File "D:\Application\anaconda3\envs\py36-tf\lib\site-packages\mne\datasets\utils.py", line 448, in _download
    hash_=hash_)

  File "<string>", line 2, in _fetch_file

  File "D:\Application\anaconda3\envs\py36-tf\lib\site-packages\mne\utils.py", line 952, in verbose
    return function(*args, **kwargs)

  File "D:\Application\anaconda3\envs\py36-tf\lib\site-packages\mne\utils.py", line 2293, in _fetch_file
    % (temp_file_name, hash_, md5))

RuntimeError: Hash mismatch for downloaded file C:\Users\BCI-13\mne_data\MNE-sample-data-processed.tar.gz.part, expected fc2d5b9eb0a144b1d6ba84dc3b983602 but got 12b75d1cb7df9dfb4ad73ed82f61094f
`
what shoul I do ?

drammock, 2020-08-14:
hash mismatch can indicate that the file was corrupted during download. The first thing I would try is to delete the downloaded file and try again.

drammock, 2020-08-14:
I just downloaded the sample data version 5, and I also get the hash `12b75d1cb7df9dfb4ad73ed82f61094f`.  Seems like maybe the hash we're checking against is incorrect. I'll look into it.

drammock, 2020-08-14:
actually, I'm failing to reproduce this.  Current master version is checking for the correct hash.  Could you do two things:

1. repaste your traceback [formatted as code](https://docs.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax#quoting-code).  Right now, the underscores in our function and variable names are getting interpreted as instructions to italicize parts of the traceback, which makes it harder to follow.

2. paste the output of `mne.sys_info()` (also formatted as code).  I suspect what might be going on is that your version of MNE-Python is older, and checks for the hash value of an older version of sample data but is downloading a newer version of sample data.

iuype, 2020-08-15:
> actually, I'm failing to reproduce this. Current master version is checking for the correct hash. Could you do two things:
>
> 1. repaste your traceback [formatted as [...]

Thanks，
I have changed the version of MNE from 0.17 to the stable one.
when I ues the stable version of MNE，everything goes well.

closed 2020-08-15 by iuype