mne-tools/mne-python #8120: Load NIRS data recorded using nirsLAB v15.3

DylanMannKrzisnik, 2020-08-14:
Hello MNE users,

Following recommendations from the mailing list, I'm opening an issue with regards to loading NIRS data collected from a NIRx system, using NIRScout software version 15.3. Loading data using ***mne.io.read_raw_nirx()*** works for data collected using NIRScout versions 15.0 and. 15.2, but not 15.3.

I've created a [repo](https://github.com/DylanMannKrzisnik/NIRS_sample_data) to host some sample data. In short, it consists of a finger-tapping block-design experiment with bilateral motor coverage using 8 sources and 4 detectors. As mentioned in the repo's README file:

    The following command serves to import data using MNE-Python:

    **raw_intensity = mne.io.read_raw_nirx("full_path_to_data/NIRS_sample_data")**

    However, this results in the following error message:

    **RuntimeError: MNE does not support this NIRStar version ("15.3")**

More info in the repo.

What would be the best way around this issue? If anyone has suggestions, my labmates and I can look into it. Just wanted to know if anyone has some advice. Let me know if anything is missing from the repo.

Regards,

Dylan Mann-Krzisnik

larsoner, 2020-08-14:
Have you tried first just hacking the code to allow a "15.3" string here:

https://github.com/mne-tools/mne-python/blob/master/mne/io/nirx/nirx.py#L110-L111

Try making that change then following the fNIRS preprocessing tutorial:

https://mne.tools/dev/auto_tutorials/preprocessing/plot_70_fnirs_processing.html#sphx-glr-auto-tutorials-preprocessing-plot-70-fnirs-processing-py

That will tell us if things at least mostly work already.

Then someone will need to figure out what changed between 15.2 and 15.3 to see if we actually need to add test files, etc. If they only made irrelevant changes (e.g., changing fields we don't use or adding new fields we don't care about) then modifying the string in `master` might already be good enough.

cc @rob-luke in case you have other ideas

DylanMannKrzisnik, 2020-08-14:
Hi again,

Indeed this seems to work! By adding "\"15.3\"" to the iterable in
L110-L111 of nirx.py I was able to run the tutorial scripts, with some
modifications to adapt to the sample data.
I should have done that first before posting anything. However, I was weary
about the downstream effects of such a simple modification. This alludes to
your point about needing to know what changed between 15.2 & 15.3.

We'll do more research on our side to see what the potential problems could
be. If anyone else can provide further insight, that would be great.

Thank you,

Dylan MK

drammock, 2020-08-14:
Thanks @DylanMannKrzisnik for looking into it. I'll leave this issue open until you report back whether there are any (relevant) differences between 15.2 and 15.3

DylanMannKrzisnik, 2020-08-14:
I've contacted the NIRx support team and there are no changes to the file format which are relevant to our inquiry.

This means that simply adding version 15.3 to the list of supported versions does the trick.

Best,

Dylan MK

larsoner, 2020-08-14:
Would you be up for making a pull request to add it?

DylanMannKrzisnik, 2020-08-14:
Of course, will do.

larsoner, 2020-08-24:
This was fixed by #8122

closed 2020-08-24 by larsoner