mne-tools/mne-python #8127: Retaining complex stc.data when applying a morph

ualsbombe, 2020-08-18:
If your issue is a usage question, please consider asking on our [Gitter channel](https://gitter.im/mne-tools/mne-python) or on our [mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) instead of opening an issue.

#### Describe the problem
Please provide a clear and concise description of the problem.
Morphing an stc with complex data results in losing the imaginary part

#### Describe your solution
A clear and concise description of what you want to happen.
I would like to retain the values as stc.data as complex numberts

#### Describe possible alternatives
A clear and concise description of any alternative solutions or features you have considered.

#### Additional context
Add any other context or screenshots about the feature request here.
I tried to make mne.morph._interpolate_data inherit the dtype of stc.data (picture from morph.py)
[image]

but there seem to be later parts that also drop the imaginary part. I haven't been able to identify those despite an hour's attempt or so
I get the following warning, whose origin I cannot identify

[image]

agramfort, 2020-08-18:
can you share a tiny and self-sufficient script to replicate the problem?
it helps us investigate

>

ualsbombe, 2020-08-18:
you'll get it first thing tomorrow

larsoner, 2020-08-18:
@ualsbombe can you try the fix in https://github.com/mne-tools/mne-python/pull/8129 ?

ualsbombe, 2020-08-19:
@larsoner still a little new to this. I have been trying to create a branch with the changes in your commit, but I cannot checkout the commit.
Tried `git checkout 5f0af3af7963dec7ebf138473465101f0cbb2d52`, but as it is not present in the repository, I cannot check it out plus `git log` doesn't show it. Can you please give me a pointer as I would be happy to test it out
[image]

hoechenberger, 2020-08-19:
@ualsbombe The easiest way to install @larsoner's branch is by doing:
```
pip install -U https://github.com/larsoner/mne-python/archive/complex.zip
```

Does that work for you?

ualsbombe, 2020-08-19:
That worked - thanks @hoechenberger

agramfort, 2020-08-19:
all good then?

>

ualsbombe, 2020-08-19:
testing it now - running the morph now - I am not getting any warnings. will get back when the morph is done and I have checked whether data is indeed complex (and no errors)

ualsbombe, 2020-08-19:
yup, data is complex after the morph now and no errors -will make a small script, so you can see for yourselves

ualsbombe, 2020-08-19:
here's a self-sufficient example

import mne
from os.path import join

data_path = mne.datasets.sample.data_path()
subjects_dir = join(data_path, 'subjects')
subject = 'sample'
MEG_path = join(data_path, 'MEG', subject)

raw = mne.io.read_raw(join(MEG_path, 'sample_audvis_raw.fif'), preload=True)
raw.pick_types(meg=True)
cov = mne.compute_raw_covariance(raw)
raw.filter(14, 30)
raw.apply_hilbert()
events = mne.read_events(join(MEG_path, 'sample_audvis_raw-eve.fif'))
epochs = mne.Epochs(raw, events, 1, tmin=-0.200, tmax=0.600, baseline=None)
data_cov = mne.compute_covariance(epochs)

fwd = mne.read_forward_solution(join(MEG_path,
                                     'sample_audvis-meg-vol-7-fwd.fif'))
filters = mne.beamformer.make_lcmv(epochs.info, fwd, data_cov, 0.05, cov,
                                   pick_ori='max-power', 
                                   weight_norm='unit-noise-gain')

stcs = mne.beamformer.apply_lcmv_epochs(epochs, filters)

## source space
src = fwd['src']

morph = mne.compute_source_morph(src, subject, 'fsaverage',
                                 subjects_dir=subjects_dir)

# test one epoch
stc = stcs[0]
stc_morph = morph.apply(stc)
print(stc_morph.data.dtype)

ualsbombe, 2020-08-19:
`import mne`
`from os.path import join`

`data_path = mne.datasets.sample.data_path()`
`subjects_dir = join(data_path, 'subjects')`
`subject = 'sample'`
`MEG_path = join(data_path, 'MEG', subject)`

`raw = mne.io.read_raw(join(MEG_path, 'sample_audvis_raw.fif'), preload=True)`
`raw.pick_types(meg=True)`
`cov = mne.compute_raw_covariance(raw)`
`raw.filter(14, 30)`
`raw.apply_hilbert()`
`events = mne.read_events(join(MEG_path, 'sample_audvis_raw-eve.fif'))`
`epochs = mne.Epochs(raw, events, 1, tmin=-0.200, tmax=0.600, baseline=None)`
`data_cov = mne.compute_covariance(epochs)`

`fwd = mne.read_forward_solution(join(MEG_path,
                                     'sample_audvis-meg-vol-7-fwd.fif'))`
`filters = mne.beamformer.make_lcmv(epochs.info, fwd, data_cov, 0.05, cov,
                                   pick_ori='max-power', 
                                   weight_norm='unit-noise-gain')`

`stcs = mne.beamformer.apply_lcmv_epochs(epochs, filters)`

`## source space`
`src = fwd['src']`

`morph = mne.compute_source_morph(src, subject, 'fsaverage',
                                 subjects_dir=subjects_dir)`

here's a self-sufficient example
`# test one epoch`
`stc = stcs[0]`
`stc_morph = morph.apply(stc)`
`print(stc_morph.data.dtype)`

PR #8129 (2020-08-19): MRG, BUG: Fix complex vol morph
Closes #8127
closed 2020-08-19 by agramfort