mne-tools/mne-python #8130: anonymize also sex and hand

agramfort, 2020-08-19:
anonymize with keep_his False keeps what's in: `raw.info['subject_info']`
which contains hand and sex for example.

Should we remove these info with keep_his=False or should we introduce another parameter?

cbrnr, 2020-08-19:
What does "his" mean?

larsoner, 2020-08-19:
Thoughts @bloyl ?

agramfort, 2020-08-19:
HIS = health information system

bloyl, 2020-08-20:
I don't think that handedness and sex/gender are considered PHI or
Identifing information in the US.

That being said I would remove them if `keep_his=False` to not further
complicate the API.

In general I feel like depending on the fif files to hold this type of
information is error prone and probably unreliable particularly across
sites.

agramfort, 2020-08-21:
I would look at what other software do to be consistent.

Falach, 2021-03-15:
I'll take this one

agramfort, 2021-03-16:
do we know what other software are doing?

@ftadel @robertoostenveld are brainstorm or fieldtrip removing handness and gender when anonymizing?

robertoostenveld, 2021-03-16:
FieldTrip itself does not anonymize and does not have a single tool for converting data to BIDS. Instead the user would write a script around the data2bids function (looping over subjects and sessions). So a user may do it, depending on how he feeds these two into the cfg.participants.handedness and cfg.participants.sex options.

However, I personally would not do it as the entropy in these two is just too low. 

In most cases we would code both with 2 values, which means that with 5 subjects you would already have some repetition as there can only 4 possible pairs of values. For handedness you would have 90/10% occurrence. When only considering M and F as levels for sex, you would have approximately 50/50% occurence of both.In the least probable case (a female left-hander), 1 out of 22 people will still have the same sex and handedness.  However, if you were to collect and store gender, it can become more unique.

ftadel, 2021-03-17:
In Brainstorm, we don't store (yet) any subject-related meta-data beyond ID and acquisition date. So the .tsv files generated when exporting a Brainstorm database as a BIDS dataset do not include these fields. 

However, the info could remain the in headers of the recordings. Some files can be copied as is to the BIDS export, and propagate sex/handedness info. The anonymization of the file headers is not part (yet) of the export process, it remains a separate optional step (for now).

The only BIDS-compatible file format (exported by Brainstorm, ie. excluding the EEGLAB .set) that can store this information is **FIF**.
Before the export to BIDS, the user can call this function **fiff_anonymizer.m** (from a script or from the GUI), which by default removes both sex and handedness:
https://github.com/brainstorm-tools/brainstorm3/blob/master/external/other/fiff_anonymizer.m

agramfort, 2021-03-17:
thanks a lot for the quick feedback !

>

PR #9103 (2021-03-18): [MRG] anonymize sex and hand when keep_his is false
#### Reference issue
Fixes #8130

#### What does this implement/fix?
when keep_his = false it's also anonymize hand and sex

#### Additional information
I updated the relevant tests as well
closed 2021-03-18 by agramfort