mne-tools/mne-python #8131: I want to connect the EDF data with annotations.

Han-HeyGyeong, 2020-08-19:
I have a question. The error occurs when I run the code.

Code : 

```

#import
import numpy as np
import matplotlib.pyplot as plt

import mne
from mne.datasets.sleep_physionet.age import fetch_data
from mne.time_frequency import psd_welch

from sklearn.ensemble import RandomForestClassifier
from sklearn.metrics import accuracy_score
from sklearn.metrics import confusion_matrix
from sklearn.metrics import classification_report
from sklearn.pipeline import make_pipeline
from sklearn.preprocessing import FunctionTransformer 

raw1 = mne.io.read_raw_edf('SC4822G0-PSG.edf')
annot1 = mne.read_annotations('SC4822GC-Hypnogram.edf')
mapping = {'EOG horizontal': 'eog',
           'Resp oro-nasal': 'misc',
           'EMG submental': 'misc',
           'Temp rectal': 'misc',
           'Event marker': 'misc'}
raw1.set_annotations(annot1, emit_warning=False)
raw1.set_channel_types(mapping)

raw2 = mne.io.read_raw_edf('SC4821G0-PSG.edf')
annot2 = mne.read_annotations('SC4821GC-Hypnogram.edf')
mapping = {'EOG horizontal': 'eog',
           'Resp oro-nasal': 'misc',
           'EMG submental': 'misc',
           'Temp rectal': 'misc',
           'Event marker': 'misc'}
raw2.set_annotations(annot2, emit_warning=False)
raw2.set_channel_types(mapping)

raw = mne.concatenate_raws([raw1, raw2])

```

Output:
 

> 
> ---------------------------------------------------------------------------
> ValueError                                Traceback (most recent call last)
> <ipython-input-30-1f1577643877> in <module>
> ----> 1 raw = mne.concatenate_raws([raw2, raw4])
> 
> <decorator-gen-157> in concatenate_raws(raws, preload, events_list, verbose)
> 
> D:\pyth\anaconda\lib\site-packages\mne\io\base.py in concatenate_raws(raws, preload, events_list, verbose)
>    2092         first, last = zip(*[(r.first_samp, r.last_samp) for r in raws])
>    2093         events = concatenate_events(events_list, first, last)
> -> 2094     raws[0].append(raws[1:], preload)
>    2095 
>    2096     if events_list is None:
> 
> D:\pyth\anaconda\lib\site-packages\mne\io\base.py in append(self, raws, preload)
>    1509         all_raws = [self]
>    1510         all_raws += raws
> -> 1511         _check_raw_compatibility(all_raws)
>    1512 
>    1513         # deal with preloading data first (while files are separate)
> 
> D:\pyth\anaconda\lib\site-packages\mne\io\base.py in _check_raw_compatibility(raw)
>    2045             raise ValueError('raw[%d][\'info\'][\'ch_names\'] must match' % ri)
>    2046         if not all(raw[ri]._cals == raw[0]._cals):
> -> 2047             raise ValueError('raw[%d]._cals must match' % ri)
>    2048         if len(raw[0].info['projs']) != len(raw[ri].info['projs']):
>    2049             raise ValueError('SSP projectors in raw files must be the same')
> 
> ValueError: raw[1]._cals must match
> 

I thought `concatenate_raws()` connected data like a playlist(Like a YouTube playlist, B after A !).  But only `raw_test= mne.concatenate_raws([raw2, raw2.copy()])` worked. I don't know why. Should the data be the same length? The two data have different measurement times. 

I don't know how to drop data from a raw instance. Or if there is a way to solve this problem, please let me know. 

I referred to this page(https://mne.tools/stable/auto_tutorials/sample-datasets/plot_sleep.html?highlight=sleep%20eeg).

and my system info 

> Platform:      Windows-10
> Python:        3.7.6
> 
> mne:           0.20.7
> numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
> scipy:         1.4.1
> matplotlib:    3.1.3 {backend=module://ipykernel.pylab.backend_inline}
> 
> sklearn:       0.22.1
> numba:         0.48.0 
> pandas:        1.0.1

cbrnr, 2020-08-19:
Actually, `mne.concatenate_raws([raw1, raw2])` modifies `raw1` in-place and appends `raw2` (see https://mne.tools/stable/generated/mne.concatenate_raws.html). However, the error you get indicates that the two recordings are incompatible because apparently the channel scalings/ranges are different.

Han-HeyGyeong, 2020-08-19:
Thank you for your answer. Your answer gave me confidence in the cause. So as a result of thinking about it, I do what I want using `concatenate_epochs()`. Thank you.

closed 2020-08-19 by cbrnr