mne-tools/mne-python #8133: example error

5dh15, 2020-08-20:
I'm trying to run the example code on MNE official page, which is a simple code.

But it throws a error, I don't know how to solve it.
This is error code:
ValueError: The number of FixedLocator locations (20), usually from a call to set_ticks, does not match the number of ticklabels (1).

import mne
import numpy as np
import matplotlib.pyplot as plt

data = np.random.randn(5, 1000)

info = mne.create_info(
    ch_names=['MEG1', 'MEG2', 'EEG1', 'EEG2', 'EOG'],
    ch_types=['grad', 'grad', 'eeg', 'eeg', 'eog'],
    sfreq=100
)

custom_raw = mne.io.RawArray(data, info)

scalings = {'eeg': 2, 'grad': 2, 'eog':2}

plt.yticks(['MEG1', 'MEG2', 'EEG1', 'EEG2', 'EOG'], 
            labels=5)
custom_raw.plot(n_channels=5, scalings=scalings, title='Data from arrays', block=True)

cbrnr, 2020-08-20:
This has already been reported in #8031 - you can use the development version until we make a new release.

closed 2020-08-20 by cbrnr

5dh15, 2020-08-20:
> This has already been reported in #8031 - you can use the development version until we make a new release.

Thank you for your reply!

5dh15, 2020-08-21:
@cbrnr 
I can't install the dev version of mne.
Could you tell me how to install it?

Here is my install error:

C:\Users\USER>pip install mne==0.21.dev0
ERROR: Could not find a version that satisfies the requirement mne==0.21.dev0 (from versions: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.7.1, 0.8, 0.8.1, 0.8.5, 0.8.6, 0.9.0, 0.10.0, 0.11.0, 0.12.0, 0.13, 0.13.1, 0.13.2, 0.14, 0.14.1, 0.15, 0.15.1, 0.15.2, 0.16, 0.16.1, 0.16.2, 0.17.0, 0.17.1, 0.17.2, 0.18.0, 0.18.1, 0.18.2, 0.19.0, 0.19.1, 0.19.2, 0.20.0, 0.20.1, 0.20.2, 0.20.3, 0.20.4, 0.20.5, 0.20.6, 0.20.7)
ERROR: No matching distribution found for mne==0.21.dev0

bloyl, 2020-08-21:
pip install -U https://github.com/mne-tools/mne-python/archive/master.zip

5dh15, 2020-08-21:
Thank you! Helped me a lot! @bloyl