mne-tools/mne-python #8139: Complex data cannot be saved or read (stc's and evokeds)

ualsbombe, 2020-08-21:
#### Describe the bug
When complex data stored in an stc or and evokeds object is saved, the imaginary part is lost.
I found a quick fix for saving the complex data in stc's, but when reading it in again, it is lost again

#### Steps to reproduce
```
import mne
from os.path import join
from pathlib import Path
home = str(Path.home())

data_path = mne.datasets.sample.data_path()
subjects_dir = join(data_path, 'subjects')
subject = 'sample'
MEG_path = join(data_path, 'MEG', subject)

raw = mne.io.read_raw(join(MEG_path, 'sample_audvis_raw.fif'), preload=True)
raw.pick_types(meg=True)
cov = mne.compute_raw_covariance(raw)
raw.filter(14, 30)
raw.apply_hilbert()
events = mne.read_events(join(MEG_path, 'sample_audvis_raw-eve.fif'))
epochs = mne.Epochs(raw, events, 1, tmin=-0.200, tmax=0.600, baseline=None)
evokeds = epochs.average()
data_cov = mne.compute_covariance(epochs)

fwd = mne.read_forward_solution(join(MEG_path, 
                                     'sample_audvis-meg-vol-7-fwd.fif'))
filters = mne.beamformer.make_lcmv(epochs.info, fwd, data_cov, 0.05, cov,
                                   pick_ori='max-power', 
                                   weight_norm='unit-noise-gain')

stc = mne.beamformer.apply_lcmv(evokeds, filters)

# save stc
stc.save(join(home, 'delete_me'))

# save evokeds
mne.write_evokeds(join(home, 'delete_me-ave.fif'), evokeds)
```
#### Quick fix for saving data
Change line 119 (commented) to line 120 of `_write_stc` in _source_estimate.py_
It cannot be read again though since `_read_w` of _source_estimate.py_ throws away the imaginary part as well

#### Expected results
A saved object with complex data in it

#### Actual results
[image]

[image]

#### Additional information
[image]

agramfort, 2020-08-21:
try saving as .h5

>

ualsbombe, 2020-08-21:
thanks for your suggestion
both saving and loading works in the h5 format

agramfort, 2020-08-21:
ok close then.

thanks

ualsbombe, 2020-08-21:
doesn't solve the issue for evokeds though

agramfort, 2020-08-21:
did you try saving the evoked in .h5

>

ualsbombe, 2020-08-21:
mne.write_evokeds does not have an h5 option as far as I can tell, or is there a separate function for that?

agramfort, 2020-08-21:
ok then we need to fix something...

can you share a code snippet highlighting the pb?

>

ualsbombe, 2020-08-21:
Assuming that pb means problem, the code under "steps to reproduce" allows for reproducing the problem and warning.
Let me know if I misunderstood you

ualsbombe, 2020-08-24:
I have now added my pull request, but seems like I am failing some tests (notebook, style and windows pre-pip). I must admit I cannot interpret the error codes (they are bash exit codes for which I don't have the reference)
Other than that the tests worked locally and I am pretty convinced that my proposed changes will work, but will need some help understanding the failed tests
https://github.com/mne-tools/mne-python/pull/8153

PR #8153 (2020-08-25): ENH: Better complex support for Evoked and SourceEstimate
#### Reference issue
Example: Fixes #8139 

#### What does this implement/fix?
Raises an error when complex VolSourceEstimate.data is saved to formats other .h5
Allows for _write_evokeds_ and _read_evokeds_ writing and reading complex data

#### Additional information
I haven't changed _test_evoked.py_ since I did not know whether any of the datasets had complex data to test it on
closed 2020-08-25 by larsoner